SkeletonMan03/FZEasyMarauderFlash #3: Help with esp32-Wroom

TQMatvey, 2022-10-19:
Hello, flashed the the esp32-wroom binary to my esp-wroom-32, flipper doesnt work with it, connect 5v,gnd,tx,rx, nothing, board powers on, wifi marauder cant talk to it, what is the pinout for esp32-wroom??

TQMatvey, 2022-10-19:
Basically wifi marauder firmware runs fine over serial, but not on flipper..

SkeletonMan03, 2022-10-19:
Double-check your connections and try rebooting it. If you're using 5v instead of 3v3, make sure 5v on GPIO is turned on on your Flipper Zero. If it still doesn't work, open an issue on the [Marauder Github](https://github.com/justcallmekoko/ESP32Marauder)  as this tool is just for flashing the firmware. The bin it flashes comes from that repo as well

TQMatvey, 2022-10-19:
Alright, will do, thanks

closed 2022-10-19 by TQMatvey