SkeletonMan03/FZEasyMarauderFlash #4: ESP32-S2 won't flash because of NameError

Y0seph, 2022-11-19:
[image]

SkeletonMan03, 2022-11-19:
It should be defined. That's the first check for the VID of the S2 which is shown in your lsusb output. Did you run the script as `sudo python3 EasyInstall.py`? And I assume you ran step 0 first or you wouldn't have gotten that far

Y0seph, 2022-11-19:
Step 0 is done, not I didn't run anything as sudo.
Gonna try it and I'll let you know what happens

Y0seph, 2022-11-19:
The problem persists. Just for background, I'm running it through a linux vm on my Chromebook. It is the native vm and it's registering the ESP32 as connected.

SkeletonMan03, 2022-11-19:
Could you try the latest commit and see if that helps?

Y0seph, 2022-11-19:
> Could you try the latest commit and see if that helps?

What do you mean by latest commit? Do you mean "comment"? If so, I did try, and the error message is the same 🥲
In the end Chromebooks aren't the best for learning programing

SkeletonMan03, 2022-11-19:
I meant I just updated the script, can you try the latest version?

Y0seph, 2022-11-19:
[image]
[image]

SkeletonMan03, 2022-11-19:
I've tested on Arch Linux, Debian, FreeBSD, Windows 10, Windows 11, and a Raspberry Pi since you've opened this issue and it worked on all of them. I'm honestly quite confused because I've been unable to reproduce this issue. Could you show the output of `dmesg | grep tty`?

SkeletonMan03, 2022-11-19:
After some research, I'm starting to think the ChromeOS host is not allowing the native Linux VM on your Chromebook to connect to the UART Bridge. While `lsusb` may see the device, if you don't see the device with `dmesg | grep tty` (which should show `/dev/ACM0` or `/dev/ttyUSB0`) then it may simply not be supported. Unfortunately, I don't have a Chromebook to test and attempt to reproduce this with.

Y0seph, 2022-11-19:
I have been held back from other things on Flipper because of the Chromebook, so it isn't really any surprise. 
I thank you for the help though, and if I somehow get it to work I'll let you know ;)

closed 2022-11-19 by Y0seph