SkeletonMan03/FZEasyMarauderFlash #6: Not working in Linux

vajranaga, 2022-11-27:
Without root permissions cannot access `/dev/tty*` (but the prep steps in first menus work fine). Using `sudo python EasyInstall.py` I get:

```
Traceback (most recent call last):
  File "/path/to/FZEasyMarauderFlash/EasyInstall.py", line 4, in <module>
    from git import Repo
ModuleNotFoundError: No module named 'git'
```
Tried also within venv, and with various options. Python3 and Python point to same bin for me on Fedora 37.

I have tried `sudo -E python EasyInstall.py`, `(flipper_venv) $ sudo -E /path/to/flipper_venv/bin/python EasyInstall.py` which also gives errors.

`(flipper_venv) $ sudo /path/to/flipper_venv/bin/pip install -r path/to/requirements.txt` and then trying the above, and again no joy.

Thank you for the script!

Using `zsh 5.9 (x86_64-redhat-linux-gnu)`

```
$ cat /proc/version
Linux version 6.0.9-300.fc37.x86_64 (mockbuild@bkernel01.iad2.fedoraproject.org) (gcc (GCC) 12.2.1 20220819 (Red Hat 12.2.1-2), GNU ld version 2.38-24.fc37) #1 SMP PREEMPT_DYNAMIC Wed Nov 16 17:36:22 UTC 2022
```
```
(flipper_venv) $ pip --version                   
pip 22.3.1 from /path/to/flipper_venv/lib64/python3.11/site-packages/pip (python 3.11)
```

SkeletonMan03, 2022-11-27:
It's not surprising that it won't work without root. You'd probably need to add your user to the dialout group to avoid it. You don't have the requirements installed. Install them and it should work. Specifically, you're missing gitpython, and you may need `git` to be installed as well, so if you don't have it, run `sudo yum install git`. This has worked with no problem on Debian, Arch, Ubuntu, Raspbian, FreeBSD, Mac OS, and Windows 10/11. In all Linux distros, we did not use full paths for every single line because it is unnecessary. `python3` specifically was used, not `python` because Raspbian has python2.7 as `python`, and you should install the requirements with `pip3`, but since those are symlinked in your case, that shouldn't matter. `bash` was used as the shell on all of the Linux distros I tried and for FreeBSD, but also shouldn't matter.

vajranaga, 2022-11-27:
I have git installed, the output is the same with python3 or python command (aliased like with pip and I've tried both versions for every command). I have tested with system packages, I am in dialout (and wheel) group. It also didn't work with sudo, as mentioned above. I tried it in bash, no joy.

It's a fresh install of Fedora 37.

```
(flipper_venv)  vajranaga@localhost  ~/flipper  pip list
Package      Version
------------ -------
bitstring    3.1.9
cffi         1.15.1
cryptography 38.0.3
ecdsa        0.18.0
esptool      4.4
gitdb        4.0.10
GitPython    3.1.29
glob2        0.7
pip          22.3.1
pycparser    2.21
pyserial     3.5
reedsolo     1.5.4
setuptools   62.6.0
six          1.16.0
smmap        5.0.0
```
All the required packages are also in system python.
``` vajranaga@localhost  ~/flipper/FZEasyMarauderFlash   main  sudo python3 EasyInstall.py 
Traceback (most recent call last):
  File "/home/vajranaga/flipper/FZEasyMarauderFlash/EasyInstall.py", line 4, in <module>
    from git import Repo
ModuleNotFoundError: No module named 'git'
```
Same output inside venv.

I have tried all the options you suggested before posting. I've tried even more since. I have modified the EasyInstall.py script in a few ways to test other things too (package names, etc). Still nothing.

Does not run under sudo, and cannot talk to /dev/ttyACM0 as non-root.

Thanks for the reply, but that didn't change anything.

SkeletonMan03, 2022-11-27:
I'm confused why it would be any different with Fedora than other Linux systems even if you're using a different shell and how you'd be missing `gitpython` if it were installed. Could you try installing the requirements as root, then run the script as root?

vajranaga, 2022-11-27:
Okay, running in system python, running `sudo pip install -r requirements.txt` then `sudo python3 EasyInstall.py` works.

Thank you. I was really hoping to avoid messing with system packages as root for this. It's caused me problems before building the fw for the flipper.

I'll mark it closed, but maybe keep it as a reference for anyone else trying to do it inside venv, or trying to not install system packages with root.

closed 2022-11-27 by vajranaga

SkeletonMan03, 2022-11-27:
As you said, it's not best practice to run it as root, and I'd prefer not to. It can be avoided on some systems by making sure your user is in the wheel and dialout groups, but it doesn't always work. If you have any suggestions or anything, I'm open to PRs. It has given an interesting challenge.

vajranaga, 2022-11-27:
I may just poke around and see if I can figure something out. Everything flipper has been a full stack learning experience (just a hobbyist).