SkeletonMan03/FZEasyMarauderFlash #9: Serial port identification found wrong port

andrewmeyer, 2022-12-06:
It looks like the script short circuited and accidentially selected a different device located at /dev/ttyUSB0 when the board was located at /dev/ttyACM0.

It may be possible to search /dev/serial/by-id for a device labeled usb-Espressif and identify which serial port it is using that way.
```
 /dev/serial/by-id  ll  
lrwxrwxrwx 1 root root 13 Dec  6 10:09 usb-Espressif_ESP32-S2_0-if00 -> ../../ttyACM0
lrwxrwxrwx 1 root root 13 Dec  5 14:54 usb-Silicon_Labs_CP2105_Dual_USB_to_UART_Bridge_Controller_00C9832A-if00-port0 -> ../../ttyUSB0
lrwxrwxrwx 1 root root 13 Dec  5 14:54 usb-Silicon_Labs_CP2105_Dual_USB_to_UART_Bridge_Controller_00C9832A-if01-port0 -> ../../ttyUSB1
```
```
DISTRIB_ID=LinuxMint
DISTRIB_RELEASE=20
DISTRIB_CODENAME=ulyana
DISTRIB_DESCRIPTION="Linux Mint 20 Ulyana"
```

andrewmeyer, 2022-12-06:
That was my thought too, only a Linux solution haha.  Sadly I am not well versed in windows/mac, but it may be possible that Mac's bsd roots may provide a similar enumeration feature.  the silicon_labs devices are two connections to a ham radio - I unplugged them and the board flashed seamlessly

SkeletonMan03, 2022-12-06:
Since it works fine with only the chip you're trying to flash plugged in, I'll close this for now pending a Pythonic solution to this, but I'm glad to know what happens with multiple devices with the same VIDs are plugged in. I'd rather not resort going back to doing OS detection and using os.system again.

closed 2022-12-06 by SkeletonMan03