SkeletonMan03/FZEasyMarauderFlash #10: "Since you've gotten this far, your ESP32 chip can be detected, but you do not have permissions to erase or flash it"

Cainor, 2022-12-08:
First of all, I'd like to thank you for your work.

I'm trying to use the your script to flash the my FZ WiFi dev board. But facing the issue below:

```
//======================================================\\
|| Options:                                             ||
|| 1) Flash Marauder on WiFi Devboard or ESP32-S2       ||
|| 2) Save Flipper Blackmagic WiFi settings             ||
|| 3) Flash Flipper Blackmagic                          ||
|| 4) Flash Marauder on ESP32-WROOM                     ||
|| 5) Update all files                                  ||
|| 6) Exit                                              ||
\\======================================================//

Please enter the number of your choice: 3
You have chosen to flash Flipper Blackmagic
←[34mYou are most likely using a Flipper Zero WiFi Devboard or an ESP32-S2←[0m
Flashing Flipper Blackmagic with WiFi Settings restore
Erasing firmware...
esptool.py v4.4
Serial port COM6
Connecting.....
Chip is ESP32-S2 (revision v0.0)
Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 60:55:f9:ed:5e:02
Uploading stub...
Running stub...
Stub running...
Erasing region (may be slow depending on size)...
Erase completed successfully in 0.4 seconds.
Staying in bootloader.
←[31mSomething went wrong!←[0m
←[31mSince you've gotten this far, your ESP32 chip can be detected, but you do not have permissions to erase or flash it, or you selected the wrong option←[0m
←[31mTry running the script again with sudo or as Administrator←[0m
```

I'm using a CMD shell with Administrative permissions.

SkeletonMan03, 2022-12-08:
To double-check, did you hold the boot button as you plugged it in, then press and release the reset button, then release the boot button?

SkeletonMan03, 2022-12-08:
Also, I just made a (minor) modification. Could you re-download the script and try again?

Cainor, 2022-12-08:
@SkeletonMan03, again thank you for the quick response.
Still same issue. And yes,
-Hold BOOT
-Plugin
-Press RESET
-Release BOOT

SkeletonMan03, 2022-12-08:
I removed the try block where it's getting stuck since I think esptool is erroneously throwing some exception. Could you pull the latest version and try again? Sorry about that. It worked on a WROOM fine, but I had no devboard or S2 to test with

Cainor, 2022-12-08:
Bro, don't apology .. I'd like to apology for annoying you with this issue :)

Anyway, since the "try" block was removed, I'm seeing the actual error message:
`
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: OSError(22, 'A device which does not exist was specified.', None, 433)
`

SkeletonMan03, 2022-12-08:
Ah, interesting. I've not seen that error get thrown. Can you check in device manager and make sure that you have any driver you may need for it?

Cainor, 2022-12-08:
[image]

Hmm .. Does this means I need to download the driver fro ESP32?

SkeletonMan03, 2022-12-08:
Interesting. I added a print for it to show what serial port it detected as a sanity check, but that won't resolve the issue. I'll do some testing when I get home from work, but it was working fine last time I was testing in both Linux and Windows

Cainor, 2022-12-08:
I think this is an issue from my end. Your code is fine.

Cainor, 2022-12-09:
Just wanted to update this issue. That I got it to work with the Original repo:
https://github.com/UberGuidoZ/Flipper/tree/main/Wifi_DevBoard/FZ_Marauder_Flasher

It showed the same error I faced here, but it continued to run regardless. And it worked.

Note that I tried it before and it didn't, but it seems like that the issue was my driver with ESP32-S2. It worked after I used:  Zadig: https://zadig.akeo.ie/

closed 2022-12-09 by Cainor