SkeletonMan03/FZEasyMarauderFlash #11: Input/output error on Arch

TheT3ngu, 2022-12-15:
When I try to flash the device, it seems that my OS can't communicate with the board. The output is:

`Please enter the number of your choice: 1
You have chosen to flash Marauder on a WiFi devboard or ESP32-S2
You are most likely using a Flipper Zero WiFi Devboard or an ESP32-S2
Erasing firmware...
esptool.py v4.4
Serial port /dev/ttyACM0
Connecting........
Traceback (most recent call last):
  File "/home/tengu/Schreibtisch/FZEasyMarauderFlash/EasyInstall.py", line 247, in <module>
    choose_fw()
  File "/home/tengu/Schreibtisch/FZEasyMarauderFlash/EasyInstall.py", line 93, in choose_fw
    flash_esp32marauder()
  File "/home/tengu/Schreibtisch/FZEasyMarauderFlash/EasyInstall.py", line 195, in flash_esp32marauder
    erase_esp32fw()
  File "/home/tengu/Schreibtisch/FZEasyMarauderFlash/EasyInstall.py", line 129, in erase_esp32fw
    esptool.main(['-p', serialport, '-b', BR, '-c', chip, '--before', 'default_reset', '-a', 'no_reset', 'erase_region', '0x9000', '0x6000'])
  File "/usr/lib/python3.10/site-packages/esptool/__init__.py", line 666, in main
    esp = esp or get_default_connected_device(
  File "/usr/lib/python3.10/site-packages/esptool/__init__.py", line 915, in get_default_connected_device
    _esp.connect(before, connect_attempts)
  File "/usr/lib/python3.10/site-packages/esptool/loader.py", line 632, in connect
    last_error = self._connect_attempt(
  File "/usr/lib/python3.10/site-packages/esptool/loader.py", line 556, in _connect_attempt
    self.bootloader_reset(usb_jtag_serial, extra_delay)
  File "/usr/lib/python3.10/site-packages/esptool/loader.py", line 537, in bootloader_reset
    self._setDTR(False)  # IO0=HIGH, done
  File "/usr/lib/python3.10/site-packages/esptool/loader.py", line 450, in _setDTR
    self._port.setDTR(state)
  File "/usr/lib/python3.10/site-packages/serial/serialutil.py", line 603, in setDTR
    self.dtr = value
  File "/usr/lib/python3.10/site-packages/serial/serialutil.py", line 473, in dtr
    self._update_dtr_state()
  File "/usr/lib/python3.10/site-packages/serial/serialposix.py", line 715, in _update_dtr_state
    fcntl.ioctl(self.fd, TIOCMBIC, TIOCM_DTR_str)
OSError: [Errno 5] Input/output error
`
Sorry, this is my first ESP-board and I have no clue where the error lies exactly. I tried to flash both with and without `sudo` and getting the same error. Also, my user is in the `dialout`-group.

SkeletonMan03, 2022-12-15:
I've not gotten an Input/Output error while flashing one before. All I can tell you for certain is this isn't an Arch problem as I've been using it on Arch and those errors appear to be coming from esptool.
Are you running this on an ESP32-S2 or a WiFi Devboard?
Did you put the chip into bootloader mlde by doing the following?
1) Connect the devboard or ESP32 board via USB.
2) Press and hold the BOOT button on the module, press and release the RESET button.
3) Release the BOOT button.

TheT3ngu, 2022-12-16:
Thaks for the response. I tested it on another system (also Arch), getting the same error there.
It's a WiFi devboard, the sticker on the front reads: "Flipper Zero Wi-Fi Module v1". Sticker on the back states that it contains a ESPS2WR0VER. I tried the following combinations to get the board into download mode:

1. Connect the board via USB, hold BOOT, press RESET, release BOOT
2. Hold BOOT, connect the board via USB, release BOOT
3. Hold BOOT, connect the board via USB, press RESET, release BOOT
4. Connect the board, hold BOOT for 3 seconds

Nothing of it works, still getting the same error. May my board be broken, anything else I could try?

SkeletonMan03, 2022-12-16:
Wrover? The Flipper devboard has an ESP32-S2 on it, but don't believe it's the Wrover, can you post a picture of the board where the actual chip is visible?

SkeletonMan03, 2022-12-16:
Could you also send the output of `lsusb`?

SkeletonMan03, 2022-12-16:
Cool, everything else looks good, so I'm fairly certain your esptool install is broken somehow. Not your board, or even my tool. Could you try `pip3 install --upgrade --no-deps --force-reinstall esptool`?

TheT3ngu, 2022-12-16:
You were right! I still don't know what exactly causes the problem on my Arch, but I ran your installer successfull after setting it up in an Ubuntu live enviroment, where it completed the flash in 6 seconds. Everything works now, thank you so much for your time and help!

closed 2022-12-16 by TheT3ngu

SkeletonMan03, 2022-12-16:
I'm curious how that was broken on 2 Arch installs for you, but not on either of my Arch installs, we must have set our stuff up differently. I'm glad you got it working.

seanhagen, 2023-07-08:
I had the same thing happen, and the `esptool` re-install got things kind of working. I say "kind of" because during the erasing/flashing process the dev board got reset and then showed up as `/tty/ACM1` instead of `/tty/ACM0`, causing the tool to fail:

```
$ python3 EasyInstall.py                                                                                                         1521ms  Sat 08 Jul 2023 08:05:27 PM
                                                                                                                                                                                                         
#########################################################################                                                                                                                                
#    Marauder Flasher Script                                            #                                                                                                                                
#    Python edition by SkeletonMan based off of a Windows Batch         #                                                                                                                                
#    script by Frog, UberGuidoz, and ImprovingRigamarole                #                                                                                                                                
#                                                                       #                                                                                                                                
#    Thanks to everyone who has done testing on various chips for me    #                                                                                                                                
#    Thanks to Scorp for compiling needed bins for the ESP32-WROOM      #                                                                                                                                
#########################################################################                                                                                                                                
                                                                                                    
Make sure your ESP32 or WiFi devboard is plugged in!                                                                                                                                                     
Checking for prerequisites...                                                                                                                                                                            
The extra ESP32 bins folder exists!                                                                                                                                                                      
Checking for Marauder releases                                                                                                                                                                           
Great, you have the Marauder releases folder!                                                                                                                                                            
ESP32 Marauder firmware exists at ESP32Marauder/releases/esp32_marauder_v0_10_8_20230621_flipper.bin 
ESP32 Marauder firmware exists at ESP32Marauder/releases/esp32_marauder_v0_10_8_20230621_flipper_sd_serial.bin                                                                                           
ESP32-S3 firmware bin exists at ESP32Marauder/releases/esp32_marauder_v0_10_8_20230621_multiboardS3.bin                                                                                                  
Old Hardware bin exists at ESP32Marauder/releases/esp32_marauder_v0_10_8_20230621_old_hardware.bin                                                                                                       
Mini bin exists at ESP32Marauder/releases/esp32_marauder_v0_10_8_20230621_mini.bin                                                                                                                       
                                            
//======================================================\\                                                                                                                                               
|| Options:                                             ||                                          
|| 1) Flash Marauder on WiFi Devboard or ESP32-S2       ||                                                                                                                                               
|| 2) Flash SD Serial Marauder on Devboard or ESP32-S2  ||                                                                                                                                               
|| 3) Save Flipper Blackmagic WiFi settings             ||                                                                                                                                               
|| 4) Flash Flipper Blackmagic                          ||                                          
[... cut ...]
WARNING: ESP32-S2 (revision v0.0) chip was placed into download mode using GPIO0.                                                                                                                        
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.                                                                                                             
To suppress this note, set --after option to 'no_reset'.                             
```

Not sure what happened, but figured it's worth reporting here so you have another data point for comparison!