SkeletonMan03/FZEasyMarauderFlash #12: Flipper Marauder firmware version now exceeds pattern matching

ieatlint, 2022-12-23:
It looks like due to the firmware updating beyond the range checked,  line [152](https://github.com/SkeletonMan03/FZEasyMarauderFlash/blob/main/EasyInstall.py#L152) this line needs to be updated from:
`esp32marauderfwc=('ESP32Marauder/releases/esp32_marauder_v[0-9]_[0-9]_[0-9][0-9]_*_flipper.bin')`

The current firmware version as I type this is v0.10.0 or:
`esp32_marauder_v0_10_0_20221222_flipper.bin`

This failed to match the "10_0" part as it's expecting it be "N_NN" instead of "NN_N".

Because of this, I was getting this error:
`NameError: name 'esp32marauderfw' is not defined`

By modifying line 152 to read this, I was able to continue:
`esp32marauderfwc=('ESP32Marauder/releases/esp32_marauder_v[0-9]_[0-9][0-9]_[0-9]_*_flipper.bin')`

However this isn't backwards compatible with earlier firmware, and will break again when v0.10.10 (or later) is released, so this is purely a hack and not a solution.  Likely a regex solution will need to be pursued, or this would be in the form of a PR.

SkeletonMan03, 2022-12-23:
I was just wondering about this right before you submitted this issue. I'm working on it now. I think the only solution to avoid name changes like that, temporarily at least, may be for me to remove all of the [0-9]s and just use *

SkeletonMan03, 2022-12-23:
I have fixed the issue with the latest commit, it should work for now. While I don't like just using *, it avoids issues with changes like this

closed 2022-12-23 by SkeletonMan03