SkeletonMan03/FZEasyMarauderFlash #13: Can’t use marauder with flipper app . . . Only get gibberish.

Kirball, 2022-12-28:
I have successfully flashed my esp32 wroom (x3). They work fine in standalone boards or when I TTL into them.  The problem is when I use one of my boards connect via gpio and run the marauder app through the flipper. Anything I do just returns gibberish on the screen. If I start a scan I get random strings on nonsense and this happens for anything I try to do in the app (including the reboot option).   Do you know what causes this or how I can fix it?

SkeletonMan03, 2022-12-28:
I've seen that happen before, try pressing the reset button on your chip

Kirball, 2022-12-28:
I have tried that with no luck,   It's happening with both of my flippers on ofw and cfw.  Uberz suggested I track you down last night to ask about it figured this was the quickest easiest way to contact you.

SkeletonMan03, 2022-12-28:
I'd try re-flashing and make sure you're connecting it properly. I've only seen that happen once with an ESP32-S3, but pressing reset fixed it for me

Kirball, 2022-12-28:
Yeah, I've tried that as well with no luck.

SkeletonMan03, 2022-12-28:
How did you connect your ESP32-WROOM to your Flipper Zero? I'm not really sure why it would be doing that.

Kirball, 2022-12-28:
https://imgur.com/a/m5dbTA3 and via direct wires from ESP to GPIO.

SkeletonMan03, 2022-12-29:
I would suggest trying to connect it via jumper wires to see if it works that way, but it would be a pain and your soldering looks fine from what I can tell. Maybe double-check the Tx0 to Rx connection, but I'm honestly not sure what's up with it.

Kirball, 2022-12-29:
I have also tried that as well. I have 3 flashed esps and just got 3 more today. I have tried just about everything I can think of or that has been suggested. I have tried on 2 different flippers one with ofw and one with cfw. I can’t for the life of me figure out what’s going on. I wonder if I should flash an older version as I am using 0.10.0

SkeletonMan03, 2022-12-29:
Sorry for the late reply, I've been pretty busy. You could try an older version or something, but I have no idea why it wouldn't work, so I don't think I'd be much help to you. I'd suggest joining the Marauder Discord for assistance at this point because this is something that isn't just related to FZEasyMarauderFlash.

closed 2022-12-29 by SkeletonMan03