SkeletonMan03/FZEasyMarauderFlash #16: dlsym(RTLD_DEFAULT, kIOMainPortDefault): symbol not found

HappyZ, 2023-01-10:
Got the following error on OSX (Big Sur)
```
python3 EasyInstall.py
Traceback (most recent call last):
  File "EasyInstall.py", line 8, in <module>
    import serial.tools.list_ports
  File "/Users/happyz/anaconda3/lib/python3.6/site-packages/serial/tools/list_ports.py", line 29, in <module>
    from serial.tools.list_ports_posix import comports
  File "/Users/happyz/anaconda3/lib/python3.6/site-packages/serial/tools/list_ports_posix.py", line 31, in <module>
    from serial.tools.list_ports_osx import comports
  File "/Users/happyz/anaconda3/lib/python3.6/site-packages/serial/tools/list_ports_osx.py", line 32, in <module>
    kIOMasterPortDefault = ctypes.c_void_p.in_dll(iokit, "kIOMainPortDefault")
ValueError: dlsym(RTLD_DEFAULT, kIOMainPortDefault): symbol not found
```

HappyZ, 2023-01-10:
Changed to kIOMainPortDefault and not working either. Thought it was because kIOMasterPortDefault being deprecated.

SkeletonMan03, 2023-01-13:
Since you're using Anaconda instead of a normal install of Python and Pip, I'm not sure it would even work. I'd suggest not using Anaconda as this tool wasn't meant to be used with it

closed 2023-01-17 by SkeletonMan03