SkeletonMan03/FZEasyMarauderFlash #18: Issue writing to Wemos ESP32 S2 Mini

jduraes, 2023-01-13:
Hello,

As per the subject, I have a little Wemos S2 Mini that I'm using to make a small DYI board.

When trying to write the firmware to it using the generic ESP32-S2 option 1, I get the following error trace:

```
Please enter the number of your choice: 1
You have chosen to flash Marauder on a WiFi devboard or ESP32-S2
You are most likely using a Flipper Zero WiFi Devboard or an ESP32-S2
Erasing firmware...
esptool.py v4.4
Serial port COM12
Connecting....
Chip is ESP32-S2FNR2 (revision v0.0)
Features: WiFi, Embedded Flash 4MB, Embedded PSRAM 2MB, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 48:27:e2:4d:f1:4a
Uploading stub...
Running stub...
Stub running...
Erasing region (may be slow depending on size)...
Erase completed successfully in 0.2 seconds.
Staying in bootloader.
Traceback (most recent call last):
  File "C:\Users\username\Documents\github\FZEasyMarauderFlash\EasyInstall.py", line 249, in <module>
    choose_fw()
  File "C:\Users\username\Documents\github\FZEasyMarauderFlash\EasyInstall.py", line 95, in choose_fw
    flash_esp32marauder()
  File "C:\Users\username\Documents\github\FZEasyMarauderFlash\EasyInstall.py", line 197, in flash_esp32marauder
    erase_esp32fw()
  File "C:\Users\username\Documents\github\FZEasyMarauderFlash\EasyInstall.py", line 131, in erase_esp32fw
    esptool.main(['-p', serialport, '-b', BR, '-c', chip, '--before', 'default_reset', '-a', 'no_reset', 'erase_region', '0x9000', '0x6000'])
  File "C:\Program Files\Python39\lib\site-packages\esptool\__init__.py", line 849, in main
    esp.soft_reset(True)  # exit stub back to ROM loader
  File "C:\Program Files\Python39\lib\site-packages\esptool\loader.py", line 1399, in soft_reset
    self.flash_finish(True)
  File "C:\Program Files\Python39\lib\site-packages\esptool\loader.py", line 857, in flash_finish
    self.check_command("leave Flash mode", self.ESP_FLASH_END, pkt)
  File "C:\Program Files\Python39\lib\site-packages\esptool\loader.py", line 407, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "C:\Program Files\Python39\lib\site-packages\esptool\loader.py", line 394, in command
    self._port.timeout = saved_timeout
  File "C:\Program Files\Python39\lib\site-packages\serial\serialutil.py", line 372, in timeout
    self._reconfigure_port()
  File "C:\Program Files\Python39\lib\site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
    raise SerialException(
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31)
```

I can write other images and Arduino sketches to this board using the same COM12 (putting the device into bootloader mode) just fine.

Would you know what could be causing that "PermissionError" above?

Thanks.

SkeletonMan03, 2023-01-13:
Did you try running it as admin? I've not seen that happen. I'm not sure if that specific chip even works with Marauder

jduraes, 2023-01-13:
Thanks for responding so quickly.

Yes I've also tried is administrator - same result. If you look at the
start of the transaction, it sees the COM12 port, connects and gets the
board details fine. It also appears to erase the chip.

The Mini is based on the ESP32 S2 so it should at least program it. Whether
it would work, is a different story, especially because there aren't
dedicated UART pins exposed on the board.

That would be my next step, trying to scope out what pins to use to connect
to the Flipper.

I'll keep digging. If you could leave this issue open for a few days
please, in case others comment, that would be great.

Cheers.

SkeletonMan03, 2023-01-13:
Yeah, it's quite strange that it wouldn't flash, but would erase it. I'd only expect it not to flash if it were no longer in bootloader mode, but it shouldn't be possible to even leave bootloader mode after being erased

Hecfromthebronx, 2023-05-02:
Getting the error below. I am running powershell as administrator. I right clicked and clicked 'run as...'

Please enter the number of your choice: 2
You have chosen to flash Marauder on a WiFi devboard or ESP32-S2 with support for SD Serial Support
←[34mYou are most likely using a Flipper Zero WiFi Devboard or an ESP32-S2←[0m
Erasing firmware...
esptool.py v4.5.1
Serial port COM5
Connecting...
Chip is ESP32-S2 (revision v0.0)
Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 60:55:f9:ec:2e:ce
Uploading stub...
Running stub...
Stub running...
Erasing region (may be slow depending on size)...
Erase completed successfully in 0.4 seconds.
Staying in bootloader.
Traceback (most recent call last):
  File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 308, in <module>
    choose_fw()
  File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 103, in choose_fw
    checkforserialport()
  File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 49, in checkforserialport
    choose_fw()
  File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 103, in choose_fw
    checkforserialport()
  File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 49, in checkforserialport
    choose_fw()
  File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 103, in choose_fw
    checkforserialport()
  File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 49, in checkforserialport
    choose_fw()
  File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 103, in choose_fw
    checkforserialport()
  File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 49, in checkforserialport
    choose_fw()
  File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 103, in choose_fw
    checkforserialport()
  File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 49, in checkforserialport
    choose_fw()
  File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 103, in choose_fw
    checkforserialport()
  File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 49, in checkforserialport
    choose_fw()
  File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 104, in choose_fw
    flash_esp32marauderserial()
  File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 243, in flash_esp32marauderserial
    erase_esp32fw()
  File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 145, in erase_esp32fw
    esptool.main(['-p', serialport, '-b', BR, '-c', chip, '--before', 'default_reset', '-a', 'no_reset', 'erase_region', '0x9000', '0x6000'])
  File "C:\Users\hdiff\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\esptool\__init__.py", line 855, in main
    esp.soft_reset(True)  # exit stub back to ROM loader
    ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\hdiff\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\esptool\loader.py", line 1436, in soft_reset
    self.flash_finish(True)
  File "C:\Users\hdiff\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\esptool\loader.py", line 872, in flash_finish
    self.check_command("leave Flash mode", self.ESP_FLASH_END, pkt)
  File "C:\Users\hdiff\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\esptool\loader.py", line 435, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\hdiff\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\esptool\loader.py", line 422, in command
    self._port.timeout = saved_timeout
    ^^^^^^^^^^^^^^^^^^
  File "C:\Users\hdiff\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\serial\serialutil.py", line 372, in timeout
    self._reconfigure_port()
  File "C:\Users\hdiff\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
    raise SerialException(
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: OSError(22, 'A device which does not exist was specified.', None, 433)

closed 2023-05-15 by SkeletonMan03