SkeletonMan03/FZEasyMarauderFlash #23: Issue Writing To Wifi DevBoard Model WD.1

QuePID, 2023-02-21:
Here is a log of the errors I receive:

Using Python 3.11.2 for Windows 11

D:\FZEasyMarauderFlash-main>python EasyInstall.py

#########################################
#    Marauder Flasher Script            #
#    Python edition by SkeletonMan      #
#    based off of a Windows based       #
#    script by Frog, UberGuidoz,        #
#    and ImprovingRigamarole            #
#                                       #
#    Thanks to everyone who has done    #
#    testing on various chips for me    #
#    Thanks to Scorp for compiling      #
#    needed bins for the ESP32-WROOM    #
#########################################

Make sure your ESP32 or WiFi devboard is plugged in!
Checking for prerequisites...
The extra ESP32 bins folder exists!
Checking for Marauder releases
Great, you have the Marauder releases folder!
ESP32 Marauder firmware exists at ESP32Marauder/releases\esp32_marauder_v0_10_1_20230117_flipper.bin
ESP32-S3 firmware bin exists at ESP32Marauder/releases\esp32_marauder_v0_10_1_20230117_mutliboardS3.bin
Old Hardware bin exists at ESP32Marauder/releases\esp32_marauder_v0_10_1_20230117_old_hardware.bin

//======================================================\\
|| Options:                                             ||
|| 1) Flash Marauder on WiFi Devboard or ESP32-S2       ||
|| 2) Save Flipper Blackmagic WiFi settings             ||
|| 3) Flash Flipper Blackmagic                          ||
|| 4) Flash Marauder on ESP32-WROOM                     ||
|| 5) Flash Marauder on ESP32-S3                        ||
|| 6) Update all files                                  ||
|| 7) Exit                                              ||
\\======================================================//

Please enter the number of your choice: 1
You have chosen to flash Marauder on a WiFi devboard or ESP32-S2
You are most likely using a Flipper Zero WiFi Devboard or an ESP32-S2
Erasing firmware...
esptool.py v4.4
Serial port COM13
Connecting...
Chip is ESP32-S2 (revision v0.0)
Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: a0:76:4e:84:bd:80
Uploading stub...
Running stub...
Stub running...
Erasing region (may be slow depending on size)...
Erase completed successfully in 0.4 seconds.
Staying in bootloader.
Traceback (most recent call last):
  File "D:\FZEasyMarauderFlash-main\EasyInstall.py", line 249, in <module>
    choose_fw()
  File "D:\FZEasyMarauderFlash-main\EasyInstall.py", line 95, in choose_fw
    flash_esp32marauder()
  File "D:\FZEasyMarauderFlash-main\EasyInstall.py", line 197, in flash_esp32marauder
    erase_esp32fw()
  File "D:\FZEasyMarauderFlash-main\EasyInstall.py", line 131, in erase_esp32fw
    esptool.main(['-p', serialport, '-b', BR, '-c', chip, '--before', 'default_reset', '-a', 'no_reset', 'erase_region', '0x9000', '0x6000'])
  File "C:\Users\xxx\AppData\Local\Programs\Python\Python311\Lib\site-packages\esptool\__init__.py", line 849, in main
    esp.soft_reset(True)  # exit stub back to ROM loader
    ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\xxx\AppData\Local\Programs\Python\Python311\Lib\site-packages\esptool\loader.py", line 1399, in soft_reset
    self.flash_finish(True)
  File "C:\Users\xxx\AppData\Local\Programs\Python\Python311\Lib\site-packages\esptool\loader.py", line 857, in flash_finish
    self.check_command("leave Flash mode", self.ESP_FLASH_END, pkt)
  File "C:\Users\xxx\AppData\Local\Programs\Python\Python311\Lib\site-packages\esptool\loader.py", line 407, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\xxx\AppData\Local\Programs\Python\Python311\Lib\site-packages\esptool\loader.py", line 394, in command
    self._port.timeout = saved_timeout
    ^^^^^^^^^^^^^^^^^^
  File "C:\Users\xxx\AppData\Local\Programs\Python\Python311\Lib\site-packages\serial\serialutil.py", line 372, in timeout
    self._reconfigure_port()
  File "C:\Users\xxx\AppData\Local\Programs\Python\Python311\Lib\site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
    raise SerialException(
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31)

southsko, 2023-03-19:
Same here. 

I just flashed a WROOM and it worked flawlessly, but my official flipper wifi dev board is not liking it.

kaunix, 2023-04-09:
I'm getting a similar error trying to flask my devboard:

Please enter the number of your choice: 1
You have chosen to flash Marauder on a WiFi devboard or ESP32-S2
You are most likely using a Flipper Zero WiFi Devboard or an ESP32-S2
Erasing firmware...
esptool.py v4.5.1
Serial port /dev/ttyACM0
Traceback (most recent call last):
  File "/home/os2mac/.local/lib/python3.10/site-packages/serial/serialposix.py", line 322, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
PermissionError: [Errno 13] Permission denied: '/dev/ttyACM0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/os2mac/.local/lib/python3.10/site-packages/esptool/loader.py", line 302, in __init__
    self._port = serial.serial_for_url(port)
  File "/home/os2mac/.local/lib/python3.10/site-packages/serial/__init__.py", line 90, in serial_for_url
    instance.open()
  File "/home/os2mac/.local/lib/python3.10/site-packages/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 13] could not open port /dev/ttyACM0: [Errno 13] Permission denied: '/dev/ttyACM0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/os2mac/marauder/FZEasyMarauderFlash/EasyInstall.py", line 308, in <module>
    choose_fw()
  File "/home/os2mac/marauder/FZEasyMarauderFlash/EasyInstall.py", line 99, in choose_fw
    flash_esp32marauder()
  File "/home/os2mac/marauder/FZEasyMarauderFlash/EasyInstall.py", line 235, in flash_esp32marauder
    erase_esp32fw()
  File "/home/os2mac/marauder/FZEasyMarauderFlash/EasyInstall.py", line 145, in erase_esp32fw
    esptool.main(['-p', serialport, '-b', BR, '-c', chip, '--before', 'default_reset', '-a', 'no_reset', 'erase_region', '0x9000', '0x6000'])
  File "/home/os2mac/.local/lib/python3.10/site-packages/esptool/__init__.py", line 674, in main
    esp = esp or get_default_connected_device(
  File "/home/os2mac/.local/lib/python3.10/site-packages/esptool/__init__.py", line 920, in get_default_connected_device
    _esp = chip_class(each_port, initial_baud, trace)
  File "/home/os2mac/.local/lib/python3.10/site-packages/esptool/loader.py", line 304, in __init__
    raise FatalError(f"Could not open {port}, the port doesn't exist")
esptool.util.FatalError: Could not open /dev/ttyACM0, the port doesn't exist

Wh1t3Fox, 2023-04-12:
@kaunix you're getting permission denied errors. Try with sudo.

I'm having issues, but with an OSError

closed 2023-08-29 by SkeletonMan03