SkeletonMan03/FZEasyMarauderFlash #26: Unable to flash firmware - bootloader mode at device /dev/cu.usbmodem01 instead of cu.usbserial-1430

sysfu, 2023-04-16:
Any troubleshooting ideas for how to resolve this? Board currently has some bricked version of wifi marauder on it.

When I put the board into bootloader mode, by holding boot and pressing and releasing reset, the ESP32-S2 USB interface appears at device  /dev/cu.usbmodem01

```
Please enter the number of your choice: 1
You have chosen to flash Marauder on a WiFi devboard or ESP32-S2
You are most likely using a knock-off ESP32 chip! Success is not guaranteed!
Erasing firmware...
esptool.py v4.5.1
Serial port /dev/cu.usbserial-1430
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/serial/serialposix.py", line 322, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 16] Resource busy: '/dev/cu.usbserial-1430'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/esptool/loader.py", line 302, in __init__
    self._port = serial.serial_for_url(port)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/serial/__init__.py", line 90, in serial_for_url
    instance.open()
  File "/usr/local/lib/python3.11/site-packages/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 16] could not open port /dev/cu.usbserial-1430: [Errno 16] Resource busy: '/dev/cu.usbserial-1430'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/seth/Downloads/EasyInstall.py", line 308, in <module>
    choose_fw()
  File "/Users/seth/Downloads/EasyInstall.py", line 99, in choose_fw
    flash_esp32marauder()
  File "/Users/seth/Downloads/EasyInstall.py", line 235, in flash_esp32marauder
    erase_esp32fw()
  File "/Users/seth/Downloads/EasyInstall.py", line 145, in erase_esp32fw
    esptool.main(['-p', serialport, '-b', BR, '-c', chip, '--before', 'default_reset', '-a', 'no_reset', 'erase_region', '0x9000', '0x6000'])
  File "/usr/local/lib/python3.11/site-packages/esptool/__init__.py", line 674, in main
    esp = esp or get_default_connected_device(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/esptool/__init__.py", line 920, in get_default_connected_device
    _esp = chip_class(each_port, initial_baud, trace)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/esptool/loader.py", line 304, in __init__
    raise FatalError(f"Could not open {port}, the port doesn't exist")
esptool.util.FatalError: Could not open /dev/cu.usbserial-1430, the port doesn't exist
```

```
$ ~/Downloads [1]> ls -ltr /dev/cu.*
crw-rw-rw-  1 root  wheel   18,   1 Apr 16 12:51 /dev/cu.wchusbserial1430
crw-rw-rw-  1 root  wheel   18,   5 Apr 16 12:51 /dev/cu.MPTube-GAIA
crw-rw-rw-  1 root  wheel   18,   7 Apr 16 12:51 /dev/cu.FamblyiPad-WirelessiAP
crw-rw-rw-  1 root  wheel   18,   3 Apr 16 12:51 /dev/cu.Bluetooth-Incoming-Port
crw-rw-rw-  1 root  wheel   18,   9 Apr 16 12:51 /dev/cu.usbserial-1430
crw-rw-rw-  1 root  wheel   18,  11 Apr 16 12:53 /dev/cu.usbmodem01
```

SkeletonMan03, 2023-04-16:
Since it's detecting it as a knockoff chip, it's not guaranteed to work. But holding boot the entire time seems to work with knockoff chips

sysfu, 2023-04-16:
This is the official flipper wifi dev board, ordered from their web site, so that would be strange to me if it was a in fact a knockoff chip.

I'll try holding down the boot button the entire time, thx.

Earlier I had flashed the board using this command.

```
esptool.py -p /dev/cu.usbmodem01 -b 460800 --before default_reset --after hard_reset --chip esp32s2 write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x1000 bootloader.bin 0x10000 blackmagic.bin 0x8000 partition-table.bin
pip3 install esptool
```

sysfu, 2023-04-16:
Tried holding boot button down, inserting USB cable, then running script, same error.

SkeletonMan03, 2023-04-16:
Can you run lsusb (or the equivalent for your OS) to check its VID and PID?

SkeletonMan03, 2023-04-16:
You may also need to run it with sudo

sysfu, 2023-04-16:
During the course of determining the VID and PID, I solved the problem. 

For some weird reason, the device was accessible and writeable via esptool.py when connected by the keyboard USB hub. When I went looking for the VID and PID using Apple System information it did not appear.

So I plugged it into a USB port at the back of my iMac. Now the VID and PID are appearing, like so.

```
ESP32-S2:

  Product ID:	0x0002
  Vendor ID:	0x303a
  Version:	7.23
  Serial Number:	0
  Speed:	Up to 12 Mb/s
  Manufacturer:	Espressif
  Location ID:	0x14100000 / 7
  Current Available (mA):	500
  Current Required (mA):	500
  Extra Operating Current (mA):	0

```

Ran `python3 EasyInstall.py` again and now it's working!

Thank you for your quick response and help, and for making this software, very helpful.

closed 2023-04-16 by sysfu