SkeletonMan03/FZEasyMarauderFlash #27: esptool.util.FatalError: Failed to connect to ESP32-S2: No serial data received.

thisplantguy, 2023-04-26:
I already completed the steps up until running the EasyInstall.py script, and everything seemed fine. My Wi-Fi Devboards LED is a solid Red color, but following your steps doesn't seem to get it out of that mode and/or error.

I choose option 1, Flash Marauder on WiFi Devboard or ESP32-S2, and the script begins and seems to work, until a few second in where I get the error message shown in the title.
H
Here are the last 3 lines I get in the script:

` File "/Library/Python/3.9/site-packages/esptool/loader.py", line 647, in connect
    raise FatalError(
esptool.util.FatalError: Failed to connect to ESP32-S2: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
`

If it means anything, I just got my Flipper Zero in one of the latest batches along with the Wi-Fi board, specifically on the PCB it states Flipper Zero Wi-Fi Module V1.  I'm stuck at this point. I'm using a CalDigit TS3 Thunderbolt hub and my front-facing USB-A port with my flipper cable, and have never had any problems with the port, cable, or power delivery so far.

SkeletonMan03, 2023-04-28:
Did you run the script with `sudo`? Can you see the device when you run `system_profiler -detailLevel full`? It should have a Vendor ID of 303a if it is an ESP32-S2

thisplantguy, 2023-04-28:
When I ran system_profiler, I saw a number of instances of the ESP-32, here is one example:

`  | +-o Blackmagic ESP32@02100000  <class IOUSBHostDevice, id 0x10000d035, registered, matched, active, busy 0 (252 ms), retain 34>
    | |   |           |       |   | {
    | |   |           |       |   |   "sessionID" = 0x17b6c96505d
    | |   |           |       |   |   "USBSpeed" = 0x1
    | |   |           |       |   |   "idProduct" = 0x4001
    | |   |           |       |   |   "iManufacturer" = 0x1
    | |   |           |       |   |   "bDeviceClass" = 0xef
    | |   |           |       |   |   "IOPowerManagement" = {"PowerOverrideOn"=Yes,"DevicePowerState"=0x2,"CurrentPowerState"=0x2,"CapabilityFlags"=0x8000,"MaxPowerState"=0x2,"DriverPowerState"=0x0}
    | |   |           |       |   |   "bcdDevice" = 0x100
    | |   |           |       |   |   "bMaxPacketSize0" = 0x40
    | |   |           |       |   |   "iProduct" = 0x2
    | |   |           |       |   |   "iSerialNumber" = 0x3
    | |   |           |       |   |   "bNumConfigurations" = 0x1
    | |   |           |       |   |   "UsbDeviceSignature" = <3a3001400001626c61636b6d61676963ef02010202000a0000>
    | |   |           |       |   |   "USB Product Name" = "Blackmagic ESP32"
    | |   |           |       |   |   "locationID" = 0x2100000
    | |   |           |       |   |   "bDeviceSubClass" = 0x2
    | |   |           |       |   |   "bcdUSB" = 0x200
    | |   |           |       |   |   "kUSBSerialNumberString" = "blackmagic"
    | |   |           |       |   |   "USB Address" = 0x8
    | |   |           |       |   |   "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBHostFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
    | |   |           |       |   |   "kUSBCurrentConfiguration" = 0x1
    | |   |           |       |   |   "bDeviceProtocol" = 0x1
    | |   |           |       |   |   "USBPortType" = 0x0
    | |   |           |       |   |   "IOServiceDEXTEntitlements" = (("com.apple.developer.driverkit.transport.usb"))
    | |   |           |       |   |   "USB Vendor Name" = "Flipper Devices Inc."
    | |   |           |       |   |   "Device Speed" = 0x1
    | |   |           |       |   |   "idVendor" = 0x303a
    | |   |           |       |   |   "kUSBProductString" = "Blackmagic ESP32"
    | |   |           |       |   |   "USB Serial Number" = "blackmagic"
    | |   |           |       |   |   "IOGeneralInterest" = "IOCommand is not serializable"
    | |   |           |       |   |   "kUSBAddress" = 0x8
    | |   |           |       |   |   "kUSBVendorString" = "Flipper Devices Inc."`

thisplantguy, 2023-04-28:
Sorry... it was a lot of output. After searching through the entire output of that command, I got this:

```
Blackmagic ESP32:

          Product ID: 0x4001
          Vendor ID: 0x303a
          Version: 1.00
          Serial Number: blackmagic
          Speed: Up to 12 Mb/s
          Manufacturer: Flipper Devices Inc.
          Location ID: 0x02100000 / 8
          Current Available (mA): 500
          Current Required (mA): 100
          Extra Operating Current (mA): 0
```

thisplantguy, 2023-04-28:
Sorry--- for some reason unbeknownst to me, even when I am in the correct folder in Terminal now I'm getting this same sort of nonsense:

```
Erasing firmware...
esptool.py v4.5.1
Serial port /dev/cu.usbmodemblackmagic3
Connecting......................................
Traceback (most recent call last):
  File "/Users/REDACTED/Documents/GitHub/FZEasyMarauderFlash/EasyInstall.py", line 308, in <module>
    choose_fw()
  File "/Users/REDACTED/Documents/GitHub/FZEasyMarauderFlash/EasyInstall.py", line 99, in choose_fw
    flash_esp32marauder()
  File "/Users/REDACTED/Documents/GitHub/FZEasyMarauderFlash/EasyInstall.py", line 235, in flash_esp32marauder
    erase_esp32fw()
  File "/Users/REDACTED/Documents/GitHub/FZEasyMarauderFlash/EasyInstall.py", line 145, in erase_esp32fw
    esptool.main(['-p', serialport, '-b', BR, '-c', chip, '--before', 'default_reset', '-a', 'no_reset', 'erase_region', '0x9000', '0x6000'])
  File "/Library/Python/3.9/site-packages/esptool/__init__.py", line 674, in main
    esp = esp or get_default_connected_device(
  File "/Library/Python/3.9/site-packages/esptool/__init__.py", line 921, in get_default_connected_device
    _esp.connect(before, connect_attempts)
  File "/Library/Python/3.9/site-packages/esptool/loader.py", line 647, in connect
    raise FatalError(
esptool.util.FatalError: Failed to connect to ESP32-S2: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

I redacted my username on here after pasting the terminal contents, but everything else is the same.

SkeletonMan03, 2023-04-28:
Okay, so it can see it, but it's not getting any data from it. It doesn't appear to be in bootloader mode

SkeletonMan03, 2023-04-28:
Hold boot on the board as you are plugging it in via USB, then press reset, then release the boot button and try running it again

thisplantguy, 2023-04-28:
Hey SkeletonMan03,

I tried doing what you said; I held down boot as I was plugging it in via
USB, pressed reset, the color of the LED changed to purple/white for a
moment, then went to solid red again. I tried re-running the script and get
the same exact error.

SkeletonMan03, 2023-04-28:
Try doing that again, but holding boot the entire time as you run it. That seems to work for some people

thisplantguy, 2023-04-28:
Sigh... perhaps there's something wrong with the hardware on this board, because now no matter what I try, including your latest suggestion, I'm getting this:

```
sudo python3 EasyInstall.py

#########################################
#    Marauder Flasher Script		#
#    Python edition by SkeletonMan	#
#    based off of a Windows Batch	#
#    script by Frog, UberGuidoz,	#
#    and ImprovingRigamarole		#
#					#
#    Thanks to everyone who has done	#
#    testing on various chips for me	#
#    Thanks to Scorp for compiling	#
#    needed bins for the ESP32-WROOM	#
#########################################

Make sure your ESP32 or WiFi devboard is plugged in!
Checking for prerequisites...
The extra ESP32 bins folder exists!
Checking for Marauder releases
Great, you have the Marauder releases folder!
ESP32 Marauder firmware exists at ESP32Marauder/releases/esp32_marauder_v0_10_3_20230414_flipper.bin
ESP32 Marauder firmware exists at ESP32Marauder/releases/esp32_marauder_v0_10_3_20230414_flipper_sd_serial.bin
ESP32-S3 firmware bin exists at ESP32Marauder/releases/esp32_marauder_v0_10_3_20230414_mutliboardS3.bin
Old Hardware bin exists at ESP32Marauder/releases/esp32_marauder_v0_10_3_20230414_old_hardware.bin
Mini bin exists at ESP32Marauder/releases/esp32_marauder_v0_10_3_20230414_mini.bin

//======================================================\\ 
|| Options:						||
|| 1) Flash Marauder on WiFi Devboard or ESP32-S2	||
|| 2) Flash SD Serial Marauder on Devboard or ESP32-S2	||
|| 3) Save Flipper Blackmagic WiFi settings		||
|| 4) Flash Flipper Blackmagic				||
|| 5) Flash Marauder on ESP32-WROOM			||
|| 6) Flash Marauder on ESP32 Wemos D1 Mini		||
|| 7) Flash Marauder on ESP32-S3			||
|| 8) Update all files					||
|| 9) Exit						||
\\======================================================//

Please enter the number of your choice: 1
You have chosen to flash Marauder on a WiFi devboard or ESP32-S2
You are most likely using a Flipper Zero WiFi Devboard or an ESP32-S2
Erasing firmware...
esptool.py v4.5.1
Serial port /dev/cu.usbmodemblackmagic1
Connecting......................................
Traceback (most recent call last):
  File "/Users/martywasserman/Documents/GitHub/FZEasyMarauderFlash/EasyInstall.py", line 308, in <module>
    choose_fw()
  File "/Users/martywasserman/Documents/GitHub/FZEasyMarauderFlash/EasyInstall.py", line 99, in choose_fw
    flash_esp32marauder()
  File "/Users/martywasserman/Documents/GitHub/FZEasyMarauderFlash/EasyInstall.py", line 235, in flash_esp32marauder
    erase_esp32fw()
  File "/Users/martywasserman/Documents/GitHub/FZEasyMarauderFlash/EasyInstall.py", line 145, in erase_esp32fw
    esptool.main(['-p', serialport, '-b', BR, '-c', chip, '--before', 'default_reset', '-a', 'no_reset', 'erase_region', '0x9000', '0x6000'])
  File "/Library/Python/3.9/site-packages/esptool/__init__.py", line 674, in main
    esp = esp or get_default_connected_device(
  File "/Library/Python/3.9/site-packages/esptool/__init__.py", line 921, in get_default_connected_device
    _esp.connect(before, connect_attempts)
  File "/Library/Python/3.9/site-packages/esptool/loader.py", line 647, in connect
    raise FatalError(
esptool.util.FatalError: Failed to connect to ESP32-S2: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
martywasserman@MoeckBook-Air-2 FZEasyMarauderFlash % 

```

SkeletonMan03, 2023-04-28:
You're not getting into bootloader mode somehow. Keep trying. The serial port should come up as `/dev/cu.usbmodem01`, but your's is `/dev/cu.usbmodemblackmagic1` showing that it has booted into Blackmagic

SkeletonMan03, 2023-05-04:
> Hello, I have the same trouble but on the port `/dev/ttyACM0` on Ubuntu 23.04. Here the traceback:
> 
> ```
> //======================================================\\ 
> || Options:						||
> || 1) Flash Marauder on WiFi Devboard or ESP32-S2	||
> || 2) Flash SD Serial Marauder on Devboard or ESP32-S2	||
> || 3) Save Flipper Blackmagic WiFi settings		||
> || 4) Flash Flipper Blackmagic				||
> || 5) Flash Marauder on ESP32-WROOM			||
> || 6) Flash Marauder on ESP32 Wemos D1 Mini		||
> || 7) Flash Marauder on ESP32-S3			||
> || 8) Update all files					||
> || 9) Exit						||
> \\======================================================//
> 
> Please enter the number of your choice: 1
> You have chosen to flash Marauder on a WiFi devboard or ESP32-S2
> You are most likely using a Flipper Zero WiFi Devboard or an ESP32-S2
> Erasing firmware...
> esptool.py v4.5.1
> Serial port /dev/ttyACM0
> Connecting......................................
> Traceback (most recent call last):
>   File "/home/lionel/Git/FZEasyMarauderFlash/EasyInstall.py", line 308, in <module>
>     choose_fw()
>   File "/home/lionel/Git/FZEasyMarauderFlash/EasyInstall.py", line 98, in choose_fw
>     checkforserialport()
>   File "/home/lionel/Git/FZEasyMarauderFlash/EasyInstall.py", line 49, in checkforserialport
>     choose_fw()
>   File "/home/lionel/Git/FZEasyMarauderFlash/EasyInstall.py", line 99, in choose_fw
>     flash_esp32marauder()
>   File "/home/lionel/Git/FZEasyMarauderFlash/EasyInstall.py", line 235, in flash_esp32marauder
>     erase_esp32fw()
>   File "/home/lionel/Git/FZEasyMarauderFlash/EasyInstall.py", line 145, in erase_esp32fw
>     esptool.main(['-p', serialport, '-b', BR, '-c', chip, '--before', 'default_reset', '-a', 'no_reset', 'erase_region', '0x9000', '0x6000'])
>   File "/usr/local/lib/python3.11/dist-packages/esptool/__init__.py", line 674, in main
>     esp = esp or get_default_connected_device(
>                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/usr/local/lib/python3.11/dist-packages/esptool/__init__.py", line 921, in get_default_connected_device
>     _esp.connect(before, connect_attempts)
>   File "/usr/local/lib/python3.11/dist-packages/esptool/loader.py", line 647, in connect
>     raise FatalError(
> esptool.util.FatalError: Failed to connect to ESP32-S2: No serial data received.
> For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
> ```

This looks a bit different. Did you try running it with `sudo`?

ILPlais, 2023-05-05:
> This looks a bit different. Did you try running it with `sudo`?

Yes I did. And I did the same to install the Python requirements, so they will be able for the user `root` and not only for the local user.

thisplantguy, 2023-05-14:
I'm honestly not sure what I did differently this time, but this time there was no lit LED when I tried to run the EasyInstall.py command.

Here's the output-- I genuinely don't know if it worked or not, given the last line, but here you go

martywasserman@MoeckBook-Air-2 FZEasyMarauderFlash % sudo python3 EasyInstall.py         

#########################################
#    Marauder Flasher Script		#
#    Python edition by SkeletonMan	#
#    based off of a Windows Batch	#
#    script by Frog, UberGuidoz,	#
#    and ImprovingRigamarole		#
#					#
#    Thanks to everyone who has done	#
#    testing on various chips for me	#
#    Thanks to Scorp for compiling	#
#    needed bins for the ESP32-WROOM	#
#########################################

Make sure your ESP32 or WiFi devboard is plugged in!
Checking for prerequisites...
The extra ESP32 bins folder exists!
Checking for Marauder releases
Great, you have the Marauder releases folder!
ESP32 Marauder firmware exists at ESP32Marauder/releases/esp32_marauder_v0_10_3_20230414_flipper.bin
ESP32 Marauder firmware exists at ESP32Marauder/releases/esp32_marauder_v0_10_3_20230414_flipper_sd_serial.bin
ESP32-S3 firmware bin exists at ESP32Marauder/releases/esp32_marauder_v0_10_3_20230414_mutliboardS3.bin
Old Hardware bin exists at ESP32Marauder/releases/esp32_marauder_v0_10_3_20230414_old_hardware.bin
Mini bin exists at ESP32Marauder/releases/esp32_marauder_v0_10_3_20230414_mini.bin

//======================================================\\ 
|| Options:						||
|| 1) Flash Marauder on WiFi Devboard or ESP32-S2	||
|| 2) Flash SD Serial Marauder on Devboard or ESP32-S2	||
|| 3) Save Flipper Blackmagic WiFi settings		||
|| 4) Flash Flipper Blackmagic				||
|| 5) Flash Marauder on ESP32-WROOM			||
|| 6) Flash Marauder on ESP32 Wemos D1 Mini		||
|| 7) Flash Marauder on ESP32-S3			||
|| 8) Update all files					||
|| 9) Exit						||
\\======================================================//

Please enter the number of your choice: 1
You have chosen to flash Marauder on a WiFi devboard or ESP32-S2
You are most likely using a Flipper Zero WiFi Devboard or an ESP32-S2
Erasing firmware...
esptool.py v4.5.1
Serial port /dev/cu.usbmodem01
Connecting....
Chip is ESP32-S2 (revision v0.0)
Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 68:67:25:c0:09:fe
Uploading stub...
Running stub...
Stub running...
Erasing region (may be slow depending on size)...
Erase completed successfully in 0.3 seconds.
Staying in bootloader.
Firmware erased!
Waiting 5 seconds...
Flashing ESP32 Marauder Firmware on a WiFi Devboard or ESP32-S2...
esptool.py v4.5.1
Serial port /dev/cu.usbmodem01
Connecting......
Chip is ESP32-S2 (revision v0.0)
Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 68:67:25:c0:09:fe
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00004fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x000e2fff...
Compressed 13648 bytes to 9382...
Wrote 13648 bytes (9382 compressed) at 0x00001000 in 0.3 seconds (effective 389.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.1 seconds (effective 361.8 kbit/s)...
Hash of data verified.
Compressed 863072 bytes to 568708...
Wrote 863072 bytes (568708 compressed) at 0x00010000 in 6.1 seconds (effective 1128.3 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.
ESP32-S2 has been flashed with Marauder!
Invalid option!
martywasserman@MoeckBook-Air-2 FZEasyMarauderFlash %

SkeletonMan03, 2023-05-14:
> I'm honestly not sure what I did differently this time, but this time there was no lit LED when I tried to run the EasyInstall.py command.
>
> Here's the output-- I genuinely don't know if it worked or [...]

Seems pretty obvious that it flashed. And you should have plugged it into your FZ or connected to it via serial to check. I'm not sure how you made it throw the invalid option error, especially after it had finished.

closed 2023-05-14 by SkeletonMan03