SkeletonMan03/FZEasyMarauderFlash #28: "Invalid option!"?

Meatwad650, 2023-04-28:
The "Invalid option!" at the end shows up in red.  What option was invalid?  I'm using the Wi-Fi dev board.

`sudo python3 EasyInstall.py
Password:

#########################################
#    Marauder Flasher Script		#
#    Python edition by SkeletonMan	#
#    based off of a Windows Batch	#
#    script by Frog, UberGuidoz,	#
#    and ImprovingRigamarole		#
#					#
#    Thanks to everyone who has done	#
#    testing on various chips for me	#
#    Thanks to Scorp for compiling	#
#    needed bins for the ESP32-WROOM	#
#########################################

Make sure your ESP32 or WiFi devboard is plugged in!
Checking for prerequisites...
The extra ESP32 bins folder exists!
Checking for Marauder releases
Great, you have the Marauder releases folder!
ESP32 Marauder firmware exists at ESP32Marauder/releases/esp32_marauder_v0_10_3_20230414_flipper.bin
ESP32 Marauder firmware exists at ESP32Marauder/releases/esp32_marauder_v0_10_3_20230414_flipper_sd_serial.bin
ESP32-S3 firmware bin exists at ESP32Marauder/releases/esp32_marauder_v0_10_3_20230414_mutliboardS3.bin
Old Hardware bin exists at ESP32Marauder/releases/esp32_marauder_v0_10_3_20230414_old_hardware.bin
Mini bin exists at ESP32Marauder/releases/esp32_marauder_v0_10_3_20230414_mini.bin

//======================================================\\ 
|| Options:						||
|| 1) Flash Marauder on WiFi Devboard or ESP32-S2	||
|| 2) Flash SD Serial Marauder on Devboard or ESP32-S2	||
|| 3) Save Flipper Blackmagic WiFi settings		||
|| 4) Flash Flipper Blackmagic				||
|| 5) Flash Marauder on ESP32-WROOM			||
|| 6) Flash Marauder on ESP32 Wemos D1 Mini		||
|| 7) Flash Marauder on ESP32-S3			||
|| 8) Update all files					||
|| 9) Exit						||
\\======================================================//

Please enter the number of your choice: 1
You have chosen to flash Marauder on a WiFi devboard or ESP32-S2
You are most likely using a Flipper Zero WiFi Devboard or an ESP32-S2
Erasing firmware...
esptool.py v4.5.1
Serial port /dev/cu.usbmodem01
Connecting...
Chip is ESP32-S2 (revision v0.0)
Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 68:b6:b3:04:75:f2
Uploading stub...
Running stub...
Stub running...
Erasing region (may be slow depending on size)...
Erase completed successfully in 0.3 seconds.
Staying in bootloader.
Firmware erased!
Waiting 5 seconds...
Flashing ESP32 Marauder Firmware on a WiFi Devboard or ESP32-S2...
esptool.py v4.5.1
Serial port /dev/cu.usbmodem01
Connecting...
Chip is ESP32-S2 (revision v0.0)
Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 68:b6:b3:04:75:f2
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00004fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x000e2fff...
Compressed 13648 bytes to 9382...
Wrote 13648 bytes (9382 compressed) at 0x00001000 in 0.2 seconds (effective 451.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.1 seconds (effective 361.9 kbit/s)...
Hash of data verified.
Compressed 863072 bytes to 568708...
Wrote 863072 bytes (568708 compressed) at 0x00010000 in 6.3 seconds (effective 1099.5 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.
ESP32-S2 has been flashed with Marauder!
Invalid option!
`

SkeletonMan03, 2023-04-28:
I'm not sure what you did to cause that, but it flashed, so...

closed 2023-04-28 by SkeletonMan03

sysfu, 2023-05-21:
> The "Invalid option!" at the end shows up in red. What option was invalid? I'm using the Wi-Fi dev board.

I was able to reproduce the same error after choosing option #1 in the flash script.

```
<snip>...
Hash of data verified.

Leaving...
Staying in bootloader.
ESP32-S2 has been flashed with Marauder!
Invalid option!
user@mac /u/l/s/FZEasyMarauderFlash (main)> 
```

Environment:

macOS 10.15.7
Python 3.11.3
fish shell 3.6.1
Official Flipper WiFi dev board