SkeletonMan03/FZEasyMarauderFlash #29: Looks like it erases, but doesn't flash...

Hecfromthebronx, 2023-05-02:
Thank you for your detailed instructions. Getting the error below. I am running powershell as administrator. I right clicked and clicked 'run as...'

Please enter the number of your choice: 2
You have chosen to flash Marauder on a WiFi devboard or ESP32-S2 with support for SD Serial Support
←[34mYou are most likely using a Flipper Zero WiFi Devboard or an ESP32-S2←[0m
Erasing firmware...
esptool.py v4.5.1
Serial port COM5
Connecting...
Chip is ESP32-S2 (revision v0.0)
Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 60:55:f9:ec:2e:ce
Uploading stub...
Running stub...
Stub running...
Erasing region (may be slow depending on size)...
Erase completed successfully in 0.4 seconds.
Staying in bootloader.
Traceback (most recent call last):
File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 308, in
choose_fw()
File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 103, in choose_fw
checkforserialport()
File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 49, in checkforserialport
choose_fw()
File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 103, in choose_fw
checkforserialport()
File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 49, in checkforserialport
choose_fw()
File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 103, in choose_fw
checkforserialport()
File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 49, in checkforserialport
choose_fw()
File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 103, in choose_fw
checkforserialport()
File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 49, in checkforserialport
choose_fw()
File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 103, in choose_fw
checkforserialport()
File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 49, in checkforserialport
choose_fw()
File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 103, in choose_fw
checkforserialport()
File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 49, in checkforserialport
choose_fw()
File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 104, in choose_fw
flash_esp32marauderserial()
File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 243, in flash_esp32marauderserial
erase_esp32fw()
File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 145, in erase_esp32fw
esptool.main(['-p', serialport, '-b', BR, '-c', chip, '--before', 'default_reset', '-a', 'no_reset', 'erase_region', '0x9000', '0x6000'])
File "C:\Users\hdiff\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\esptool_init_.py", line 855, in main
esp.soft_reset(True) # exit stub back to ROM loader
^^^^^^^^^^^^^^^^^^^^
File "C:\Users\hdiff\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\esptool\loader.py", line 1436, in soft_reset
self.flash_finish(True)
File "C:\Users\hdiff\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\esptool\loader.py", line 872, in flash_finish
self.check_command("leave Flash mode", self.ESP_FLASH_END, pkt)
File "C:\Users\hdiff\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\esptool\loader.py", line 435, in check_command
val, data = self.command(op, data, chk, timeout=timeout)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\hdiff\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\esptool\loader.py", line 422, in command
self._port.timeout = saved_timeout
^^^^^^^^^^^^^^^^^^
File "C:\Users\hdiff\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\serial\serialutil.py", line 372, in timeout
self._reconfigure_port()
File "C:\Users\hdiff\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
raise SerialException(
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: OSError(22, 'A device which does not exist was specified.', None, 433)

Hecfromthebronx, 2023-05-03:
FYI.... I am able to flash the non SD version of Marauder using the flash.bat by Frog

MakeTotalSense, 2023-05-03:
I have the very same issue

SkeletonMan03, 2023-05-03:
It seems like a weird thing with Windows. I'm not 100% sure why it's happening since I can't replicate it and has worked fine for a lot of people, but I have a few ideas. Can one of you check in Computer Management if the COM port changes to something different after it erases?

MakeTotalSense, 2023-05-03:
When first plugged, COM5 and COM6 appears.
After holding BOOT, RESET, they disappear and COM7 is popping.
After running the script and getting the error, I can hear the plug (or unplug?) windows sound, the computer management refresh and display COM7. I guess it's because the device is still in 'flash me' mode

SkeletonMan03, 2023-05-03:
Okay, good, so it's staying in bootloader mode (which it should) and the COM port is staying the same. But it's timing out for some reason. After work, I'll see if I can implement a check to retry if it fails to continue after erasing. The 5 seconds it gives in between may not be enough for some systems. At the very least, I should be able to create a workaround to bypass erasing it in cases like this

SkeletonMan03, 2023-05-03:
It shouldn't in theory, but it's worth a shot

SkeletonMan03, 2023-05-03:
I think it's just taking Windows too long to refresh it when the board reboots

Hecfromthebronx, 2023-05-03:
I tried with both types of cables (USB-C on both sides and USB-C-> USB-B. Neither worked on two Windows 11 PC's I have.

SkeletonMan03, 2023-05-03:
It's going to be a while until I get home to try to implement a solution, but if you comment out erase_esp32fw() on line 243 by putting a `#` in front of it, it will skip the erase function for option 2 to force it to just flash for now without erasing first.

Hecfromthebronx, 2023-05-04:
Thanks for the quick response. Just commented out line 243 and got a little further but i don't think it fully flashed. Please see below.
______________________________________________________________________________________________________________________________

Flashing ESP32 Marauder Firmware with SD Serial support on a WiFi Devboard or ESP32-S2...
esptool.py v4.5.1
Serial port COM5
Connecting...
Chip is ESP32-S2 (revision v0.0)
Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 60:55:f9:ec:2e:ce
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00004fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x000d9fff...
Compressed 13648 bytes to 9382...
Wrote 13648 bytes (9382 compressed) at 0x00001000 in 0.3 seconds (effective 323.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.1 seconds (effective 331.8 kbit/s)...
Hash of data verified.
Compressed 826544 bytes to 540450...
Wrote 826544 bytes (540450 compressed) at 0x00010000 in 11.7 seconds (effective 565.2 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.
Traceback (most recent call last):
  File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 308, in <module>
    choose_fw()
  File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 104, in choose_fw
    flash_esp32marauderserial()
  File "C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 245, in flash_esp32marauderserial
    esptool.main(['-p', serialport, '-b', BR, '-c', chip, '--before', 'default_reset', '-a', 'no_reset', 'write_flash', '--flash_mode', 'dio', '--flash_freq', '80m', '--flash_size', '4MB', '0x1000', extraesp32bins+'/Marauder/bootloader.bin', '0x8000', extraesp32bins+'/Marauder/partitions.bin', '0x10000', esp32marauderfwserial])
  File "C:\Users\hdiff\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\esptool\__init__.py", line 855, in main
    esp.soft_reset(True)  # exit stub back to ROM loader
    ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\hdiff\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\esptool\loader.py", line 1436, in soft_reset
    self.flash_finish(True)
  File "C:\Users\hdiff\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\esptool\loader.py", line 872, in flash_finish
    self.check_command("leave Flash mode", self.ESP_FLASH_END, pkt)
  File "C:\Users\hdiff\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\esptool\loader.py", line 435, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\hdiff\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\esptool\loader.py", line 422, in command
    self._port.timeout = saved_timeout
    ^^^^^^^^^^^^^^^^^^
  File "C:\Users\hdiff\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\serial\serialutil.py", line 372, in timeout
    self._reconfigure_port()
  File "C:\Users\hdiff\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
    raise SerialException(
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: OSError(22, 'A device which does not exist was specified.', None, 433)
PS C:\Users\hdiff\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main>

Hecfromthebronx, 2023-05-04:
Update: despite the errors, it appears to have flashed ok. I did some sniffing and saw the pcaps saved on my sd card. I think we're ok. Cheers!

closed 2023-05-11 by SkeletonMan03