SkeletonMan03/FZEasyMarauderFlash #30: Sniffraw empty

itsmesikander, 2023-05-02:
When I am downloading sniffraw file to pc and dropping into wireshark. Then nothing happens its completely empty. Is there something I am missing.
Thanks

SkeletonMan03, 2023-05-11:
Check the Marauder Wiki. I'm not sure why that wouldn't work, but is not related to this tool, but Marauder itself

closed 2023-05-11 by SkeletonMan03

chrisbward, 2023-05-15:
Also having this issue when using this installer today.

Does this discussion help for a fix? 
Does this firmware installer drop the correct file in the flash process?

https://www.reddit.com/r/flipperzero/comments/13awbol/the_solution_to_the_empty_pcap_files_credit_and/

chrisbward, 2023-05-15:
Quick question - what's a known-working custom firmware to put on the flipper for this script?