SkeletonMan03/FZEasyMarauderFlash #32: NameError: name 'espoldhardwarefw' is not defined

Mr-Rog, 2023-05-04:
try to flash whit ESP32-Worm, and get this, thx for a helping hint
cheers

Codeined, 2023-05-09:
same problem... but if i try 8 then: 

Please enter the number of your choice: 8
You have chosen to update all of the files
Checking for and deleting the files before replacing them...
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 56, in choose_fw
  File "<stdin>", line 6, in update_option
FileNotFoundError: [WinError 3] The system cannot find the path specified: 'ESP32Marauder/releases'

Codeined, 2023-05-09:
bro I found the solution, just try this two steps
1.run the script as admin 
2.verify that u have installed gitforwindows

that's all, u are welcome 332

closed 2023-05-14 by SkeletonMan03