SkeletonMan03/FZEasyMarauderFlash #33: "Cannot configure port, something went wrong."

kikispcmr, 2023-05-05:
Trying to flash a flipper wifi dev board. Doesn't matter what valid option I choose, same error pops up. 
Full console output: 

`
Please enter the number of your choice: 2
You have chosen to flash Marauder on a WiFi devboard or ESP32-S2 with support for SD Serial Support
←[34mYou are most likely using a Flipper Zero WiFi Devboard or an ESP32-S2←[0m
Erasing firmware...
esptool.py v4.5.1
Serial port COM11
Connecting...
Chip is ESP32-S2 (revision v0.0)
Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 68:b6:b3:03:ca:d8
Uploading stub...
Running stub...
Stub running...
Erasing region (may be slow depending on size)...
Erase completed successfully in 0.4 seconds.
Staying in bootloader.
Traceback (most recent call last):
  File "C:\Users\kikis\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 308, in <module>
    choose_fw()
  File "C:\Users\kikis\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 104, in choose_fw
    flash_esp32marauderserial()
  File "C:\Users\kikis\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 243, in flash_esp32marauderserial
    erase_esp32fw()
  File "C:\Users\kikis\Downloads\FZEasyMarauderFlash-main\FZEasyMarauderFlash-main\EasyInstall.py", line 145, in erase_esp32fw
    esptool.main(['-p', serialport, '-b', BR, '-c', chip, '--before', 'default_reset', '-a', 'no_reset', 'erase_region', '0x9000', '0x6000'])
  File "C:\Users\kikis\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool\__init__.py", line 855, in main
    esp.soft_reset(True)  # exit stub back to ROM loader
  File "C:\Users\kikis\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool\loader.py", line 1436, in soft_reset
    self.flash_finish(True)
  File "C:\Users\kikis\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool\loader.py", line 872, in flash_finish
    self.check_command("leave Flash mode", self.ESP_FLASH_END, pkt)
  File "C:\Users\kikis\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool\loader.py", line 435, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "C:\Users\kikis\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool\loader.py", line 422, in command
    self._port.timeout = saved_timeout
  File "C:\Users\kikis\AppData\Local\Programs\Python\Python310\lib\site-packages\serial\serialutil.py", line 372, in timeout
    self._reconfigure_port()
  File "C:\Users\kikis\AppData\Local\Programs\Python\Python310\lib\site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
    raise SerialException(
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: OSError(22, 'A device which does not exist was specified.', None, 433)
`

nokozone, 2023-05-08:
Having the identical issue - any luck so far?

therealziggy, 2023-05-10:
same problem

slappomatt, 2023-05-11:
also same problem. brand new board straight from flipper.

kikispcmr, 2023-05-11:
> also same problem. brand new board straight from flipper.
> 
> EDIT: tried this one and it worked great. https://github.com/UberGuidoZ/Flipper/tree/main/Wifi_DevBoard/FZ_Marauder_Flasher

I've tried that one too but deauth doesn't seem to work tho :/ 
Not sure if they changed something with the newer dev board but I'm building my own wifi board so I'll get back to see if that's the problem.

SkeletonMan03, 2023-05-11:
Try re-downloading the tool again. I just made a change to see if it makes a difference. I think either Windows is too slow to recognize that it has restarted on some systems or they're just taking longer to restart than they should. If that doesn't fix it, I'll keep trying to replicate the issue after I get home.

therealziggy, 2023-05-11:
I switched to my Linux machine and everything worked just fine

SkeletonMan03, 2023-05-11:
I don't understand why Windows has this problem with it. And not even consistently because I've been unable to replicate it. At least it worked on Linux. I'm going to leave this issue open for now for reference.

Takun892, 2023-05-12:
It will say it erased successfully, however this will pop up every 5 seconds

SerialException: Cannot configure port, something went wrong. Original message: PermissionError(13, 'The device connected to the system is not working.', None, 31)

Used UberGuidoz to flash back to BlackMagic works fine, then flash to that version of Marauder works fine... However this tool fails to communicate once it reboots the device

SkeletonMan03, 2023-05-12:
> It will say it erased successfully, however this will pop up every 5 seconds
> 
> Please enter the number of your choice: 2 You have chosen to flash Marauder on a WiFi devboard or ESP32-S2 with SD Serial Support Checking for serial port... �[34mYou are most likely using a Flipper Zero WiFi Devboard or an ESP32-S2�[0m Erasing firmware... esptool.py v4.5.1 Serial port COM9 Connecting... Chip is ESP32-S2 (revision v0.0) Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V2 Crystal is 40MHz MAC: 68:b6:b3:04:83:40 Uploading stub... Running stub... Stub running... Erasing region (may be slow depending on size)... Erase completed successfully in 0.4 seconds. Staying in bootloader. Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31) Waiting 5 seconds and trying again... Erasing firmware... esptool.py v4.5.1 Serial port COM9 Connecting... Chip is ESP32-S2 (revision v0.0) Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V2 Crystal is 40MHz MAC: 68:b6:b3:04:83:40 Uploading stub... Running stub... Stub running... Erasing region (may be slow depending on size)... Erase completed successfully in 0.4 seconds. Staying in bootloader. Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31) Waiting 5 seconds and trying again... Erasing firmware... esptool.py v4.5.1 Serial port COM9 Connecting... Chip is ESP32-S2 (revision v0.0) Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V2 Crystal is 40MHz MAC: 68:b6:b3:04:83:40 Uploading stub... Running stub... Stub running... Erasing region (may be slow depending on size)... Erase completed successfully in 0.3 seconds. Staying in bootloader. Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31) Unable to erase the firmware on esp32s2
> 
> Used UberGuidoz tool flash back to BlackMagic works fine, then flash to that version of Marauder by justcallmekoko works fine... However this tool fails to communicate once it reboots the device

@Takun892 
What shell are you using? Did you run it as Administrator?

SkeletonMan03, 2023-05-12:
@Takun892 
I think it may be an ESPTool bug that's only affecting Windows. It appears to erase, so I'm not sure why ESPTool is throwing an error. It works exactly as intended on Linux systems I tested it on before committing  the updates last night. I'll see if I can replicate this behavior after work on Windows 11.

SkeletonMan03, 2023-05-12:
I just tested an ESP32-WROOM and an ESP32-S3-WROOM and both seem flash just fine without error on Windows 11 using Powershell without admin and as admin
Here's my full output and is what it should look like when ran in Powershell
```
←[32m
#########################################################################
#    Marauder Flasher Script                                            #
#    Python edition by SkeletonMan based off of a Windows Batch         #
#    script by Frog, UberGuidoz, and ImprovingRigamarole                #
#                                                                       #
#    Thanks to everyone who has done testing on various chips for me    #
#    Thanks to Scorp for compiling needed bins for the ESP32-WROOM      #
#########################################################################
←[0m
Make sure your ESP32 or WiFi devboard is plugged in!
Checking for prerequisites...
The extra ESP32 bins folder does not exist!
That's okay, downloading them now...
Checking for Marauder releases
Marauder releases folder does not exist, but that's okay, downloading them now...
ESP32 Marauder firmware exists at ESP32Marauder/releases\esp32_marauder_v0_10_4_20230505_flipper.bin
ESP32 Marauder firmware exists at ESP32Marauder/releases\esp32_marauder_v0_10_4_20230505_flipper_sd_serial.bin
ESP32-S3 firmware bin exists at ESP32Marauder/releases\esp32_marauder_v0_10_4_20230505_mutliboardS3.bin
Old Hardware bin exists at ESP32Marauder/releases\esp32_marauder_v0_10_4_20230505_old_hardware.bin
Mini bin exists at ESP32Marauder/releases\esp32_marauder_v0_10_4_20230505_mini.bin

//======================================================\\
|| Options:                                             ||
|| 1) Flash Marauder on WiFi Devboard or ESP32-S2       ||
|| 2) Flash SD Serial Marauder on Devboard or ESP32-S2  ||
|| 3) Save Flipper Blackmagic WiFi settings             ||
|| 4) Flash Flipper Blackmagic                          ||
|| 5) Flash Marauder on ESP32-WROOM                     ||
|| 6) Flash Marauder on ESP32 Wemos D1 Mini             ||
|| 7) Flash Marauder on ESP32-S3                        ||
|| 8) Update all files                                  ||
|| 9) Exit                                              ||
\\======================================================//

Please enter the number of your choice: 7
You have chosen to flash Marauder onto an ESP32-S3
Checking for serial port...
←[34mYou are most likely using an ESP32-WROOM, an ESP32-S2-WROVER, or an ESP32-S3-WROOM←[0m
Erasing firmware...
esptool.py v4.4
Serial port COM4
Connecting....
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 68:b6:b3:52:b3:4c
Uploading stub...
Running stub...
Stub running...
Erasing region (may be slow depending on size)...
Erase completed successfully in 0.2 seconds.
Staying in bootloader.
esp32s3 was successfully erased!
Waiting 5 seconds...
Flashing ESP32 Marauder Firmware onto ESP32-S3...
esptool.py v4.4
Serial port COM4
Connecting....
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 68:b6:b3:52:b3:4c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x000e4fff...
Warning: Image file at 0x0 is protected with a hash checksum, so not changing the flash size setting. Use the --flash_size=keep option instead of --flash_size=8MB in order to remove this warning, or use the --dont-append-digest option for the elf2image command in order to generate an image file without a hash checksum
Compressed 14896 bytes to 10242...
Wrote 14896 bytes (10242 compressed) at 0x00000000 in 1.1 seconds (effective 107.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.1 seconds (effective 332.0 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 630.1 kbit/s)...
Hash of data verified.
Compressed 868848 bytes to 566910...
Wrote 868848 bytes (566910 compressed) at 0x00010000 in -17948.0 seconds...
Hash of data verified.

Leaving...
Staying in bootloader.
←[32mESP32-S3 has been flashed with Marauder!←[0m
```

Takun892, 2023-05-12:
I'll redownload the zip and try again, not sure if it matters, but I'm on Win10 myself. If it fails i'll boot up a live linux on my laptop and try there.

Semtex5150, 2023-05-20:
Cannot configure port, something went wrong. Original message: OSError(22, 'A device which does not exist was specified.', None, 433)
Unable to erase the firmware on esp32s2

I've tried many different options with an issue with the port. Different cables as well. Any suggesestions? Thank you!

VentByte, 2023-05-21:
Same error for me. After the message `Staying in bootloader.` Windows plays the device disconnected sound and I get this error message `Cannot configure port, something went wrong. Original message: OSError(22, 'A device that does not exist was specified.', None, 433)` ( basically the same as the other folks in this thread) and the script continues to run. In total, this sound and error message come 3 times.

In the Windows Event Viewer I have 6 times the following warning: `Detected unrecognized USB driver (\Driver\USBPcap).`

I also checked the USB serial device properties in Device Manager and found that under Port Settings, the speed is set to 9600 by default. Since all the other guides I've seen regarding the WiFi Dev Board so far have always specified a speed of 115200, I changed this and tried the whole thing again, unfortunately without success. Here is a screenshot of my settings, possibly another parameter is wrong or I'm completely on the wrong track:
[image]
[image]

I also see an additional device in the device manager for which Windows did not find compatible drivers (code 28):
[image]

Also some general info about my system. Maybe this helps to reproduce the problem:
OS: Windows 11 22H2 (Build: 22621.1702)
CP210x Driver Version: v11.2.0
Python Version: 3.10.11
I'm running the script in powershell with Administrator rights.

I hope this helps to narrow down the problem a bit or to recreate it.

Semtex5150, 2023-05-21:
I also tried everything that [VentByte] discussed with the same results.

SkeletonMan03, 2023-05-22:
> Same error for me. After the message `Staying in bootloader.` Windows plays the device disconnected sound and I get this error message `Cannot configure port, something went wrong. Original message: [...]

Did you restart your computer after installing the driver?

Takun892, 2023-05-22:
I can confirm this does work on linux. It just has issues with Windows. Could be driver related, I've done restarts with my windows and it does same as their error. I'm using Win 10 22h2 as well. The linux I'm using is Kali Linux latest build on my laptop, works fine.

VentByte, 2023-05-22:
@SkeletonMan03 Yes, I have. 
I will try the whole thing with a fresh Windows 11 installation over the next few days to rule out a problem with the installation.
If that doesn't work I guess it's just thing with Windows (or like Takun892 said, with the drivers for it) device handling.

SkeletonMan03, 2023-05-22:
@VentByte Cool. Let me know what happens. Windows has caused me the most headaches with this tool.

xenowyaa, 2023-05-25:
I’m having same issues as everyone here I’ve followed all steps carefully but it just keeps saying “cannot configure port something went wrong” and it was unable to erase firmware, I’ve uninstalled git and python and reinstalled them too idk what to do now

Nvjesse, 2023-06-15:
Any updates on this issue?

BlackmanTFRC, 2023-07-31:
I have the same issue as well.
The problem is that the drivers have not been installed or are different for the latest board.
This is why everyone can see the device in the Device Manager but if you check in the device you will identify that there is not driver installed. 

I have tried couple of solutions that didn't work. So, if we find a working driver we should be able to continue flashing and working with the module. 

Any ideas?

[image]

victorkjung, 2023-08-12:
I am attempting install firmware with MacBook Pro running iOS Ventura 13.4.1 (c).
1. Instructions followed to install VCP driver of MacOsx
2. Instructions followed to attach USB - C to Wifi Dev Board
3. Red light is on the Wfi Dev board confirming connection
4. system_profiler confirms that board is recognized as Blackmagic on USB
5. Below is the error I am experiencing, which is similar to folks on this thread, but error message is slightly different.

➜  FZEasyMarauderFlash git:(main) ✗ python3 EasyInstall.py

#########################################################################
#    Marauder Flasher Script						#
#    Python edition by SkeletonMan based off of a Windows Batch		#
#    script by Frog, UberGuidoz, and ImprovingRigamarole		#
#									#
#    Thanks to everyone who has done testing on various chips for me	#
#    Thanks to Scorp for compiling needed bins for the ESP32-WROOM	#
#########################################################################

Make sure your ESP32 or WiFi devboard is plugged in!
Checking for prerequisites...
The extra ESP32 bins folder exists!
Checking for Marauder releases
Great, you have the Marauder releases folder!
ESP32 Marauder firmware exists at ESP32Marauder/releases/esp32_marauder_v0_11_0_rc3_20230727_flipper.bin
ESP32 Marauder firmware exists at ESP32Marauder/releases/esp32_marauder_v0_11_0_rc3_20230727_flipper_sd_serial.bin
ESP32-S3 firmware bin exists at ESP32Marauder/releases/esp32_marauder_v0_11_0_rc3_20230727_multiboardS3.bin
Old Hardware bin exists at ESP32Marauder/releases/esp32_marauder_v0_11_0_rc3_20230727_old_hardware.bin
Mini bin exists at ESP32Marauder/releases/esp32_marauder_v0_11_0_rc3_20230727_mini.bin

//======================================================\\
|| Options:						||
|| 1) Flash Marauder on WiFi Devboard or ESP32-S2	||
|| 2) Flash SD Serial Marauder on Devboard or ESP32-S2	||
|| 3) Save Flipper Blackmagic WiFi settings		||
|| 4) Flash Flipper Blackmagic				||
|| 5) Flash Marauder on ESP32-WROOM			||
|| 6) Flash Marauder on ESP32 Wemos D1 Mini		||
|| 7) Flash Marauder on ESP32-S3			||
|| 8) Update all files					||
|| 9) Exit						||
\\======================================================//

Please enter the number of your choice: 8
You have chosen to update all of the files
Checking for and deleting the files before replacing them...
Checking for prerequisites...
The extra ESP32 bins folder exists!
Checking for Marauder releases
Marauder releases folder does not exist, but that's okay, downloading them now...
ESP32 Marauder firmware exists at ESP32Marauder/releases/esp32_marauder_v0_11_0_rc3_20230727_flipper.bin
ESP32 Marauder firmware exists at ESP32Marauder/releases/esp32_marauder_v0_11_0_rc3_20230727_flipper_sd_serial.bin
ESP32-S3 firmware bin exists at ESP32Marauder/releases/esp32_marauder_v0_11_0_rc3_20230727_multiboardS3.bin
Old Hardware bin exists at ESP32Marauder/releases/esp32_marauder_v0_11_0_rc3_20230727_old_hardware.bin
Mini bin exists at ESP32Marauder/releases/esp32_marauder_v0_11_0_rc3_20230727_mini.bin

//======================================================\\
|| Options:						||
|| 1) Flash Marauder on WiFi Devboard or ESP32-S2	||
|| 2) Flash SD Serial Marauder on Devboard or ESP32-S2	||
|| 3) Save Flipper Blackmagic WiFi settings		||
|| 4) Flash Flipper Blackmagic				||
|| 5) Flash Marauder on ESP32-WROOM			||
|| 6) Flash Marauder on ESP32 Wemos D1 Mini		||
|| 7) Flash Marauder on ESP32-S3			||
|| 8) Update all files					||
|| 9) Exit						||
\\======================================================//

Please enter the number of your choice: 1
You have chosen to flash Marauder on a WiFi devboard or ESP32-S2
Checking for serial port...
You are most likely using a Flipper Zero WiFi Devboard or an ESP32-S2
Erasing firmware...
esptool.py v4.6.2
Serial port /dev/cu.usbmodemblackmagic3
Connecting......................................
Failed to connect to ESP32-S2: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Waiting 5 seconds and trying again...
Erasing firmware...
esptool.py v4.6.2
Serial port /dev/cu.usbmodemblackmagic3
Connecting......................................
Failed to connect to ESP32-S2: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Waiting 5 seconds and trying again...
Erasing firmware...
esptool.py v4.6.2
Serial port /dev/cu.usbmodemblackmagic3
Connecting......................................
Failed to connect to ESP32-S2: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Unable to erase the firmware on esp32s2

closed 2023-08-29 by SkeletonMan03