SkeletonMan03/FZEasyMarauderFlash #42: Serial port error: serial.tools.list_ports

quimllacer, 2023-06-11:
Running the file EasyInstall.py gives the following error:

ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found

The error comes from line 8:
`import serial.tools.list_ports`

and lines 48-52:
```
        ports=list(serial.tools.list_ports.comports())
	for vid in vids:
		for port in ports:
			if vid in port.hwid:
				serialport=port.device
				device=vid
```

These lines are used to detect the serial port (port) and vendor id (vid) of the connected ESP32.
A workaround for this error is to comment out the aforementioned lines including the following lines after line 52:
```
	serialport = "/dev/tty.usbmodem01" # Use the true serial port for your connected device
	device = "303a" # Use the true vendor id of your connected device (I believe this is the one for FZ wifi dev board
```
In summary, you must manually set the port and vid. After doing this the code worked well.

closed 2023-06-11 by SkeletonMan03

quimllacer, 2023-06-11:
> I'm not sure how you broke it, also there's a flag for specifying the port manually, it makes no sense to bother doing what you did.

Hi,

Just saying, I tried using the command to manually specify the port and it didn’t work. ProbablI won’t be able to come across this issue. I’ve read on the internet that it has to do with serial.tools.list_ports not working well after updating osx. Anyway, do what you want with this info. Your code helped me by the way, so thanks.

Have a nice one
Joaquim Llacer Wintle

SkeletonMan03, 2023-06-11:
Please send the error it gives when you use the script with
`python EasyInstall.py -s /dev/tty.usbmodem01` as using this parameter skips the entire process of checking for it and simply uses the specified port.