SkeletonMan03/FZEasyMarauderFlash #43: ESP32-WROOM-32 / Chip is ESP32-D0WD-V3 Invalid headder 0xffffffff

DRIgnazGortngschirl, 2023-06-11:
```
Please enter the number of your choice: 5
You have chosen to flash Marauder onto an ESP32-WROOM
Checking for serial port...
You are most likely using an ESP32-WROOM, an ESP32-S2-WROVER, or an ESP32-S3-WROOM
Erasing firmware...
esptool.py v4.5.1
Serial port /dev/ttyUSB0
Connecting.........
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: a0:b7:65:4e:cc:ec
```

`````

abullo1221, 2023-08-06:
With the invalid header you provided it could be these 2 issues:
1. Faulty Flashing Process: The flashing process might have been interrupted or not completed correctly. This could be due to a variety of reasons such as a loose connection, using the wrong flash size, or a power supply issue.

2. Hardware Issue: There could be a hardware issue with your ESP32 module. This could be due to a faulty module or a problem with the connection between your computer and the ESP32.

Here are some steps you can take to troubleshoot this issue:

Reflash the Firmware: Try reflashing the firmware. Make sure the connection between your computer and the ESP32 is secure and that the power supply is stable. Also, ensure that you are using the correct flash size for your device.

Check Your Hardware: If the above steps don't work, there might be a hardware issue. Check your ESP32 module for any visible signs of damage.

Update Your Flashing Tool: If you're using esptool.py, make sure it's up to date. Sometimes, updating the tool can solve the problem.

If you're still receiving the same error it's possible your Devboard is bricked. This could have been caused if you didn't follow the steps exactly to flash your devboard, or if the firmware update was interupted during the process. Or it's possible you have a faulty devboard.

DRIgnazGortngschirl, 2023-08-06:
Fixed in the meantime it was due to a faulty cable, many thanks for the response !

closed 2023-08-06 by DRIgnazGortngschirl