SkeletonMan03/FZEasyMarauderFlash #52: No Wifi(Marauder) item in the menu

Drjacky, 2023-10-05:
I used number 1 and it flashed it into FlipperZero successfully but I still don't see the Wifi(Marauder):
Mine:
[image]

Demo:
[image]

SkeletonMan03, 2023-10-05:
..........

SkeletonMan03, 2023-10-05:
This does not install the Marauder app for you on your Flipper. This installs the ESP32 FW

closed 2023-10-05 by SkeletonMan03

Drjacky, 2023-10-05:
My bad. I meant WiFi Devboard for Flipper Zero