SkeletonMan03/FZEasyMarauderFlash #54: Cannot configure port error

AlexVlah, 2023-10-10:
I have been trying to flash the wifi dev board using the option to enable the sd card and keep getting the below error. Not sure the cause I have tried several times with also wiping the board inbetween. 

`Cannot configure port, something went wrong. Original message: OSError(22, 'A device which does not exist was specified.', None, 433)`

If it would help the full log when trying to flash below. 

`Please enter the number of your choice: 2
You have chosen to flash Marauder on a WiFi devboard or ESP32-S2 with SD Serial Support
Checking for serial port...
←[34mYou are most likely using a Flipper Zero WiFi Devboard or an ESP32-S2←[0m
Erasing firmware...
esptool.py v4.6.2
Serial port COM6
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Chip is ESP32-S2 (revision v0.0)
Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 84:f7:03:93:90:c0
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 13.3s
Staying in bootloader.
Cannot configure port, something went wrong. Original message: OSError(22, 'A device which does not exist was specified.', None, 433)
Waiting 5 seconds and trying again...
Erasing firmware...
esptool.py v4.6.2
Serial port COM6
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Chip is ESP32-S2 (revision v0.0)
Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 84:f7:03:93:90:c0
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 13.3s
Staying in bootloader.
Cannot configure port, something went wrong. Original message: OSError(22, 'A device which does not exist was specified.', None, 433)
Waiting 5 seconds and trying again...
Erasing firmware...
esptool.py v4.6.2
Serial port COM6
Connecting...
Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31)
Unable to erase the firmware`

Basically it simply attempts a few times but is unsuccessful. 

If anyone has any info on this it would be greatly appreciated.

Thanks,
Alex

grafix143, 2023-10-30:
I have the same problem, @AlexVlah let me know if you found a solution.

therealziggy, 2023-11-01:
been having the same issue

Flipthermatrix, 2023-11-08:
If someone can please help me my flipper also doesn't come up on q flipper anymore and I have no idea how to flash this firmware .it. Did it after plugging in the dev. Board .any help would be appreciated

Flipthermatrix, 2023-11-08:
> been having the same issue

Hey bro did you manage to fix the issue? Did it happen to you also after plugging in dev board ?

EddieKnight1, 2023-11-10:
Had the same issue and got it solved by following the instruction for Arduino IDE.
Tips:
- Check your Windows Security notifications and allow any related to Arduino IDE. If there is any, reinstall the drivers just in case.
- Run Arduino IDE as Admin
- Ensure you choose the COM port in Arduino IDE
- Ensure you choose the Board and the Particion Scheme
- Keep Arduino open while running EasyInstall.py

Once completed, disconnect ESP32 from USB and connecting to the GPIO. 3 flashlights should occur: Light blue, green, red

Then, to test if the flashing was fine, go to the Marauder App and select SCAN. If no errors appear, the FW update completed successfully.

Flipthermatrix, 2023-11-11:
Thanks heaps my brotha!! Sorted it ..lol Rick rolled the block this morning too funny ..shits still funny…have you got any sub gig repos for Australia? Sent from my iPhone

EddieKnight1, 2023-11-11:
@AlexVlah @grafix143 @therealziggy give a try to the solution I've added above.
Also remember I'm not the developer! I can't give too much support aside of the post above.

Thumbs up are appreciated if it's solving the issue. Also will help @SkeletonMan03 as well (perhaps he updates the guide), and he can close the issue as resolved.

closed 2023-11-12 by SkeletonMan03

Flipthermatrix, 2024-03-09:
Hey how to a run black magic so I can dual boot my flipper on rougemaster? Thanks heaps any help would be great