SkeletonMan03/FZEasyMarauderFlash #55: Unable to flash blackagic: Failed to enter RAM download mode

mechgt, 2023-12-17:
I'm unable to flash the blackmagic firmware onto my ESP32-S2 chip.  I can successfully flash maurader (option 1) and evil portal (option 15), but not blackmagic. I'm connecting to the DrB0rk board through the FZ UART app.

Seems the message I'm getting is:  Failed to enter RAM download mode. Only got 2 byte status response.
... but again I don't get this when flashing marauder or evil portal.

Below is the output I get:

``` C:\...\FZEasyMarauderFlash>python EasyInstall.py
...
//==================================================================\\
|| Options:                                                         ||
||  1) Flash Marauder on WiFi Devboard or ESP32-S2                  ||
||  2) Flash SD Serial Marauder on WiFi Devboard or ESP32-S2        ||
||  3) Save Flipper Blackmagic WiFi settings                        ||
||  4) Flash Flipper Blackmagic                                     ||
||  5) Flash Marauder on ESP32-WROOM                                ||
||  6) Flash Marauder on ESP32 Marauder Mini                        ||
||  7) Flash Marauder on ESP32-S3                                   ||
||  8) Flash Marauder on AWOK v1-3 or Duoboard                      ||
||  9) Flash Marauder on AWOK v4 Chungus Board                      ||
|| 10) Flash Marauder on AWOK v5 ESP32                              ||
|| 11) Flash Marauder on AWOK Dual ESP32 (Orange Port)              ||
|| 12) Flash Marauder on AWOK Dual ESP32 Touch Screen (White Port)  ||
|| 13) Flash Marauder on AWOK Dual ESP32 Mini (White Port)          ||
|| 14) Flash Evil Portal on ESP32-WROOM                             ||
|| 15) Flash Evil Portal on ESP32-S2 or WiFi Devboard               ||
|| 16) Just Erase ESP32 - Try this if you think you bricked it      ||
|| 17) Update all files                                             ||
|| 18) Exit                                                         ||
\\==================================================================//

Please enter the number of your choice: 4
You have chosen to flash Flipper Blackmagic
Checking for serial port...
←[34mYou are most likely using an DrB0rk S3 Multiboard←[0m
Erasing firmware...
esptool.py v4.7.0
Serial port COM3
Connecting......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Chip is ESP32-S2R2 (revision v0.0)
Features: WiFi, No Embedded Flash, Embedded PSRAM 2MB, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: c0:4e:30:55:2d:2c
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 12.0s
Hard resetting via RTS pin...
Successfully erased!
Waiting 5 seconds...
Flashing Blackmagic on ESP32-S2
esptool.py v4.7.0
Serial port COM3
Connecting...
Chip is ESP32-S2R2 (revision v0.0)
Features: WiFi, No Embedded Flash, Embedded PSRAM 2MB, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: c0:4e:30:55:2d:2c
Uploading stub...
Failed to enter RAM download mode. Only got 2 byte status response.
Waiting 5 seconds and trying again...
Flashing Blackmagic on ESP32-S2
esptool.py v4.7.0
Serial port COM3
Connecting......................................
Failed to connect to ESP32-S2: Packet content transfer stopped (received 21 bytes)
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Waiting 5 seconds and trying again...
Flashing Blackmagic on ESP32-S2
esptool.py v4.7.0
Serial port COM3
Connecting.......
Chip is ESP32-S2R2 (revision v0.0)
Features: WiFi, No Embedded Flash, Embedded PSRAM 2MB, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: c0:4e:30:55:2d:2c
Uploading stub...
Failed to enter RAM download mode. Only got 2 byte status response.
Could not flash Blackmagic on ESP32-S2
```

mechgt, 2023-12-17:
Hmm.  I poked at this for several hours and consistently failed, only on Blackmagic.  Started debugging after posting this, and it just worked once. Not at all clear why.

SkeletonMan03, 2023-12-17:
Looks more like an issue with the board. I've not seen that before. Between the error and a quick Google, it looks like it's having communication issues. So it could be that you were using a bad cable

closed 2023-12-17 by SkeletonMan03