SkeletonMan03/FZEasyMarauderFlash #56: Support requested for the "cheap yellow display" ESP32-2432S028R

knightr1d3r007, 2023-12-23:
Hopefully support for the ESP32-2432S028R can be added to the scripts to build the images.
This device is ESP32, has a 2.4" display, touch screen and a SDcard reader. This hardware is practically a ready to go Marauder (well, if you bring support for the installation). Some info on this device  https://hackaday.com/2023/10/28/cheap-yellow-display-builds-community-through-hardware/
Cheers,

smoochiee, 2023-12-23:
I have one running but marauder..but touchscreen not working..

smoochiee, 2023-12-23:
https://github.com/SkeletonMan03/FZEasyMarauderFlash/assets/30816448/9d3e7235-2707-412e-abec-0156101722a1

knightr1d3r007, 2023-12-23:
Great! you were already doing work toward the same goal. I believe the issue there for the touch screen to work is that TFT_eSPI assign different pins to the touch screen. That situation is mentioned here: https://github.com/Bodmer/TFT_eSPI/discussions/2677  

I was gathering info on how to deal with that annoyance and I found this page from the youtuber LevelenteDaradici where it shows the reassignment of the pins on the file User_Setup.h for the TFT_eSPI to handle the touchscreen. Here the page: https://pastebin.com/im6ktdcx

smoochiee, 2023-12-24:
> Great! you were already doing work toward the same goal. I believe the issue there for the touch screen to work is that TFT_eSPI assign different pins to the touch screen. That situation is [...]

Already tried soldering the miso,mosi,clk of touch to lcd miso,mosi,clk..but still touch not working..added 5 way tactile button to make this usable..

knightr1d3r007, 2023-12-24:
Hey @smoochiee would you mind sharing the changes/mods you made to build the marauder for your device, so I can try your build and hopefully make the touchscreen work. I've seen you forked the Mararuder on your github, but there were no changes there.

smoochiee, 2023-12-24:
> Hey @smoochiee would you mind sharing the changes/mods you made to build the marauder for your device, so I can try your build and hopefully make the touchscreen work. I've seen you forked the Mararuder on your github, but there were no changes there.

https://github.com/smoochiee/ESP32Marauder/tree/master

smoochiee, 2023-12-26:
> Hey @smoochiee would you mind sharing the changes/mods you made to build the marauder for your device, so I can try your build and hopefully make the touchscreen work. I've seen you forked the Mararuder on your github, but there were no changes there.

Touch is working you need to solder miso,mosi,clk,and cs,,same pin as the screen..

smoochiee, 2023-12-26:
https://github.com/SkeletonMan03/FZEasyMarauderFlash/assets/30816448/0f8d193b-7a20-40b6-9fc9-92a2e43b3a2f

knightr1d3r007, 2023-12-26:
@smoochiee Would you please add details to your github repo. There are others trying the same on reddit and on the CYD on discord (https://discord.com/channels/630078152038809611/1109228361441620028/1188105082554818650). Perhaps this build can be added to the @SkeletonMan03 scripts