SkeletonMan03/FZEasyMarauderFlash #64: Crash after running marauder for

brandonfake69, 2024-04-27:
After running the wifi marauder script (or even a combination of scan / list / select / deauth) I get an error depending on how many times it cycles through the script. It also happens if the script runs the scanner for a couple of hours then selects targets, etc so it doesn't appear to be necessarily the number of times it cycles through the script it could also be how long it's been running: 

`E (5357579) wifi:Expected to init 8 rx buffer, actual is 0`
`E (5583752) wifi_init:Failed to de-init Wifi driver (0x3001)`

I bought a brand new wifi devboard and this happens on both of them with various different release of this lib. Currently running v0_13_9_20240227

brandonfake69, 2024-04-30:
I should mention this happens with both Xtreme and Unleashed firmwares

brandonfake69, 2024-05-22:
Also bought koko's wifi devboard pro and it happens on that too. Safe to say it isn't a hardware issue w/ the devboard.

SkeletonMan03, 2024-07-26:
Not an issue with this tool

closed 2024-07-26 by SkeletonMan03