SkeletonMan03/FZEasyMarauderFlash #74: Code fails when chosing ESP32-S3

nuschpl, 2024-11-06:
I'm getting:
```
Old Hardware bin exists at ESP32Marauder/releases/esp32_marauder_v1_0_0_20240626_old_hardware.bin
Mini bin exists at ESP32Marauder/releases/esp32_marauder_v1_0_0_20240626_mini.bin
New Hardware bin exists at ESP32Marauder/releases/esp32_marauder_v1_0_0_20240626_v6.bin
Checking for Evil portal
Great, you have the Evil Portal folder!
WROOM Evil Portal FW exists at EvilPortal/EvilPortalWROOM.bin
S2 Evil Portal FW exists at EvilPortal/EvilPortalS2.bin

//==================================================================\\
|| Options:                                                         ||
||  1) Flash Marauder on WiFi Devboard or ESP32-S2                  ||
||  2) Save Flipper Blackmagic WiFi settings                        ||
||  3) Flash Flipper Blackmagic                                     ||
||  4) Flash Marauder on ESP32-WROOM                                ||
||  5) Flash Marauder on ESP32 Marauder Mini                        ||
||  6) Flash v6 Marauder on ESP32-WROOM (RabbitLabs Minion Marauder)||
||  7) Flash Marauder on ESP32-S3                                   ||
||  8) Flash Marauder on AWOK v1-3 or Duoboard                      ||
||  9) Flash Marauder on AWOK v4 Chungus Board                      ||
|| 10) Flash Marauder on AWOK v5 ESP32                              ||
|| 11) Flash Marauder on AWOK Dual ESP32 (Orange Port)              ||
|| 12) Flash Marauder on AWOK Dual ESP32 Touch Screen (White Port)  ||
|| 13) Flash Marauder on AWOK Dual ESP32 Mini (White Port)          ||
|| 14) Flash Evil Portal on ESP32-WROOM                             ||
|| 15) Flash Evil Portal on ESP32-S2 or WiFi Devboard               ||
|| 16) Just Erase ESP32 - Try this if you think you bricked it      ||
|| 17) Update all files                                             ||
|| 18) Exit                                                         ||
\\==================================================================//

Please enter the number of your choice: 7
You have chosen to flash Marauder onto an ESP32-S3
Traceback (most recent call last):
  File "/home/nusch/src/FZEasyMarauderFlash/EasyInstall.py", line 584, in <module>
    choose_fw()
  File "/home/nusch/src/FZEasyMarauderFlash/EasyInstall.py", line 246, in choose_fw
    fwbin=esp32s3fw
          ^^^^^^^^^

```
and see there is some ugly global under the name **esp32s3fw**, what this code exactly tries to do ?

I'm using python3.12.4 @ venv @ linux

SkeletonMan03, 2024-11-06:
See #71

closed 2024-11-06 by SkeletonMan03

nuschpl, 2024-11-10:
Then why this platform is mimicked as supported when it's not supported ? If the upstream didn't release it might mean it's dropped it support at all. Bug is still there as this should at least fail with explicit reason